pikvm/pikvm #761: Picture-in-Picture support

berezovskyi-oleksandr, 2022-06-30:
**Is your feature request related to a problem? Please describe.**
Sometimes I don't need any control and want just to look after the output of the device
in a small window on top of all other windows.

**Describe the solution you'd like**
Would be great to have the possibility to open the video stream using Picture-in-Picture 
feature supported by modern browsers.
For example, youtube has this implemented.

mdevaev, 2022-06-30:
I don't understand how this should work if you are on other web pages.

mdevaev, 2022-07-01:
Boop

mdevaev, 2025-05-13:
There is a new API for custom picture-in-picture element: https://developer.chrome.com/docs/web-platform/document-picture-in-picture

I'll add this if I can make an arbitrarily resizable window while maintaining the proportions.