pikvm/pikvm #765: Let's Encrypt Cloudfare Documentation

realumhelp, 2022-07-01:
When running through the documentation for Let's Encrypt via Cloudflare there is a bug:

` # kvmd-pstrun -- mkdir /var/lib/kvmd/pst/data/certbot/runroot`

Should be, atleast for me none of the parent directories existed before hand.

` # kvmd-pstrun -- mkdir -p /var/lib/kvmd/pst/data/certbot/runroot`

I would also recommend linking in some credentials documentation as it had to stumble through a bit to find the right format for the credentials file.

https://certbot-dns-cloudflare.readthedocs.io/en/stable/#credentials

realumhelp, 2022-07-01:
```
[root@pikvm ~]# kvmd-pstrun -- mkdir /var/lib/kvmd/pst/data/certbot/runroot
--    INFO -- Opening PST session ...
--    INFO -- PST write is allowed: /var/lib/kvmd/pst/data
--    INFO -- Running the process ...
mkdir: cannot create directory '/var/lib/kvmd/pst/data/certbot/runroot': No such file or directory
--    INFO -- Process finished: returncode=1
```

This is why I would recommend a '-p' in the mkdir.

mdevaev, 2022-07-02:
Fixed, thanks

closed 2022-07-02 by mdevaev