pikvm/pikvm #767: kvmd-certbot.timer does not exist

realumhelp, 2022-07-02:
Just freshly downloaded and reinstalled using this image: https://files.pikvm.org/images/v3-hdmi-rpi4-latest.img.xz

When following the Let's Encrypt Docs: https://docs.pikvm.org/letsencrypt/

```
[root@pikvm ~]# systemctl enable --now kvmd-certbot.timer
Failed to enable unit: Unit file /etc/systemd/system/timers.target.wants/kvmd-certbot.timer does not exist.
```

:shrug:

_Originally posted by @realumhelp in https://github.com/pikvm/pikvm/issues/226#issuecomment-1172768042_

mdevaev, 2022-07-02:
Fixed, check the new image.

closed 2022-07-02 by mdevaev

lucaam, 2022-10-18:
> Fixed, check the new image.

I downloaded the latest image today and the situation is the same.

`Platform:
Base: Raspberry Pi 4 Model B Rev 1.4
`

`KVMD: 3.154
Streamer: 5.23 (ustreamer)
HAS_PDEATHSIG: Yes
WITH_GPIO: Yes
WITH_PTHREAD_NP: Yes
WITH_SETPROCTITLE: Yes
WITH_SYSTEMD: Yes
Linux kernel:
Machine: armv7l
Release: 5.15.68-2-rpi-ARCH
Version: #1 SMP Fri Oct 7 06:33:56 MSK 2022`

Any idea?