pikvm/pikvm #777: Recommending better VNC client for OSX and iOS

aastein, 2022-07-15:
**Is your feature request related to a problem? Please describe.**
Many VNC clients are _not good_.  

**Describe the solution you'd like**
The VNC clients by jumpdesktop are fantastic in my opinion. However they are not free. It may be worth mentioning them somewhere in the docs if that seems appropriate.

The iOS app is fantastic for VNC with a touchscreen device. I have not found any other VNC clients for iOS that work well due to the touchscreen interaction. 

iOS: https://apps.apple.com/us/app/jump-desktop-rdp-vnc-fluid/id364876095
OSX: https://apps.apple.com/us/app/jump-desktop-remote-desktop/id524141863?ls=1&mt=12

aastein, 2022-07-16:
closing because the iOS app has the windows key conversion issue but as far as i can tell it cannot be worked around

closed 2022-07-16 by aastein

mdevaev, 2022-07-16:
You can try bVNC, it's working fine.