pikvm/pikvm #780: Is the system case sensitive? gpiod

TelDragon, 2022-07-16:
sysos
```
apt-get install libgpiod-dev
pip3 install gpiod
```

err...
```
  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/aiogp.py", line 38, in <module>
    async def pulse(line: gpiod.Line, delay: float, final: float, inverted: bool=False) -> None:
AttributeError: module 'gpiod' has no attribute 'Line'. Did you mean: 'line'?

  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/aiogp.py", line 54, in <module>
    class AioReader:  # pylint: disable=too-many-instance-attributes
  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/aiogp.py", line 127, in AioReader
    def __parse_event(self, event: gpiod.LineEvent) -> Tuple[int, int]:
AttributeError: module 'gpiod' has no attribute 'LineEvent'. Did you mean: 'line_event'?

  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/plugins/atx/gpio.py", line 189, in Plugin
    async def __click(self, name: str, line: gpiod.Line, delay: float, wait: bool) -> None:
AttributeError: module 'gpiod' has no attribute 'Line'. Did you mean: 'line'?

  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/plugins/atx/gpio.py", line 200, in Plugin
    async def __inner_click(self, name: str, line: gpiod.Line, delay: float) -> None:
AttributeError: module 'gpiod' has no attribute 'Line'. Did you mean: 'line'?

  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/plugins/ugpio/gpio.py", line 88, in prepare
    self.__chip = gpiod.Chip(self.__device_path)
AttributeError: module 'gpiod' has no attribute 'Chip'. Did you mean: 'chip'?

  File "/usr/local/lib/python3.10/site-packages/kvmd-3.119-py3.10.egg/kvmd/plugins/atx/gpio.py", line 117, in sysprep
    self.__chip = gpiod.Chip(self.__device_path)
AttributeError: module 'gpiod' has no attribute 'Chip'. Did you mean: 'chip'?

    self.__power_switch_line.request("kvmd::atx::power_switch", gpiod.LINE_REQ_DIR_OUT, default_vals=[0])
AttributeError: module 'gpiod' has no attribute 'LINE_REQ_DIR_OUT'
```

After modification
```
gpiod.line
gpiod.line_event
gpiod.chip
```

mdevaev, 2022-07-16:
Everything is fine in the sources and the case is what it should be. As far as I remember, you are experimenting with Raspbian. I don't know how you managed it, but it looks like your OS is broken.

closed 2022-07-16 by mdevaev

TelDragon, 2022-07-18:
Thank you. I have found the problem.

Because the system version is too low, the gpiod module of python3 only supports gpiod cpython-37m-arm-linux-gnueabihf. so

So you need to compile and install manually

```
apt-file list python3-libgpiod

wget https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/snapshot/libgpiod-1.6.3.tar.gz
tar -xzvf libgpiod-1.6.3.tar.gz
cd libgpiod-1.6.3
./autogen.sh --enable-tools=yes
./configure --enable-tools=yes --enable-bindings-cxx=yes --enable-bindings-python=yes
make && make install
ldconfig
```