pikvm/pikvm #786: API mouse_move not effetct

lxf920125, 2022-07-27:
**Describe the bug**
I use mose_move like this:
ws.send('{"event_type": "mouse_move", "event": {"to": {"x":10000,"y":1000}}}')
however, the position of the mouse always to the top left corner, i don't know why
**To Reproduce**

**Expected behavior**
mouse to the position i set.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

mdevaev, 2022-07-27:
This cannot fail to work, this API is used by the web interface. I just tested it on my KVM with such a script. Everything is fine.
```python
import websocket, time
uri = "wss://pikvm/api/ws?stream=0"
headers = {"X-KVMD-User": "admin", "X-KVMD-Passwd": "admin"}
ws = websocket.WebSocket(sslopt={"cert_reqs": ssl.CERT_NONE})
ws.connect(uri, header=headers)
while 1:
    ws.send('{"event_type": "mouse_move", "event": {"to": {"x":10000,"y":1000}}}')
    time.sleep(1)
    ws.send('{"event_type": "mouse_move", "event": {"to": {"x":0,"y":0}}}')
    time.sleep(1)
ws.close()
```

lxf920125, 2022-07-28:
Thanks, I understood, this really works well, In fact, the position not the real position like PC screen rate, If I want to use the position in the screen, should I must remap the position like the mouse.js?
[image]
for example, the PC screen rate is 1920*1080,  I want to move to position (1200,800)

lxf920125, 2022-07-28:
Hello,
In addition, while I move the position use API, the mouse not hold on this point
the case is I will use API to control the remote PC, move mouse and click mouse left,  so I use it like this, is there other better method to operate this?
ws.send('{"event_type": "mouse_move", "event": {"to": {"x":10000,"y":1000}}}')
 ws.send('{"event_type": "mouse_button", "event": {"button":"left","state":true}}')
time.sleep(0.01)
 ws.send('{"event_type": "mouse_button", "event": {"button":"left","state":false}}')

mdevaev, 2022-07-28:
> If I want to use the position in the screen, should I must remap the position like the mouse.js?

Yes.

> while I move the position use API, the mouse not hold on this point

I don't understand. If you need something like "grab and move", you need to fire mouse_button event with true state, then move, then button again with false.

lxf920125, 2022-07-28:
thanks, this already helped for my use scenario! I already remap the position.

mdevaev, 2022-07-28:
:ok_hand:

closed 2022-07-28 by mdevaev