pikvm/pikvm #788: Harfbuzz error during upgrade

jnstockley, 2022-07-30:
**Describe the bug**
When preforming a system update harfbuzz produces error, not allowing the system to update any packages.

**To Reproduce**
Steps to reproduce the behavior, like:
1. As root run `pacman -Syu`

**Expected behavior**
Should update all system packages

**Screenshots**
Error message
```

error: failed to commit transaction (conflicting files)
harfbuzz: /usr/bin/hb-ot-shape-closure exists in filesystem
harfbuzz: /usr/bin/hb-shape exists in filesystem
harfbuzz: /usr/bin/hb-subset exists in filesystem
harfbuzz: /usr/bin/hb-view exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-aat-layout.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-aat.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-blob.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-buffer.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-common.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-cplusplus.hh exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-deprecated.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-draw.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-face.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-font.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ft.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-glib.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-gobject-enums.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-gobject-structs.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-gobject.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-graphite2.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-map.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-color.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-deprecated.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-font.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-layout.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-math.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-meta.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-metrics.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-name.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-shape.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot-var.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-ot.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-set.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-shape-plan.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-shape.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-style.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-subset-repacker.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-subset.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-unicode.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb-version.h exists in filesystem
harfbuzz: /usr/include/harfbuzz/hb.h exists in filesystem
harfbuzz: /usr/lib/cmake/harfbuzz/harfbuzz-config.cmake exists in filesystem
harfbuzz: /usr/lib/girepository-1.0/HarfBuzz-0.0.typelib exists in filesystem
harfbuzz: /usr/lib/libharfbuzz-gobject.so exists in filesystem
harfbuzz: /usr/lib/libharfbuzz-gobject.so.0 exists in filesystem
harfbuzz: /usr/lib/libharfbuzz-subset.so exists in filesystem
harfbuzz: /usr/lib/libharfbuzz-subset.so.0 exists in filesystem
harfbuzz: /usr/lib/libharfbuzz.so exists in filesystem
harfbuzz: /usr/lib/libharfbuzz.so.0 exists in filesystem
harfbuzz: /usr/lib/pkgconfig/harfbuzz-gobject.pc exists in filesystem
harfbuzz: /usr/lib/pkgconfig/harfbuzz-subset.pc exists in filesystem
harfbuzz: /usr/lib/pkgconfig/harfbuzz.pc exists in filesystem
harfbuzz: /usr/share/gir-1.0/HarfBuzz-0.0.gir exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/HarfBuzz.png exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/HarfBuzz.svg exists in filesystem
[... 94 lines cut ...]
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/utilities-common-types-apis.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/utilities.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/what-does-harfbuzz-do.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/what-harfbuzz-doesnt-do.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/what-is-harfbuzz.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/why-do-i-need-a-shaping-engine.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/why-is-it-called-harfbuzz.html exists in filesystem
harfbuzz: /usr/share/gtk-doc/html/harfbuzz/working-with-harfbuzz-clusters.html exists in filesystem
harfbuzz: /usr/share/licenses/harfbuzz/COPYING exists in filesystem
Errors occurred, no packages were upgraded.
```

**Desktop (please complete the following information):**
 - OS: iOS
 - Browser: Safari
 - Version 16
 - VNC client N/A

**PiKVM info:**
  - RPi 4
  - V3
  - PiKVM Hat
  - KVMD version: `3.127-1 `
  - uStreamer version: `5.17-1 `
  - Linux kernel: `Linux pikvm.local 5.15.34-4-rpi-ARCH #1 SMP Tue Jul 5 12:10:16 MSK 2022 armv7l GNU/Linux`

mdevaev, 2022-07-30:
Remove all listed files manually and perform update.

mdevaev, 2022-07-31:
Fixed I guess

closed 2022-07-31 by mdevaev