pikvm/pikvm #792: Bridge Discord with Matrix

JokerGermany, 2022-08-03:
Would be great if you would support Matrix (element.io), too.
Then people could would have a choice.

It would nice if you would bridge the channel with Matrix.
https://t2bot.io/discord/

mdevaev, 2022-08-04:
You can use a discord browser client if you don't want to install the desktop client. You don't even need a phone number to register. I would not like to deal with the problems of interaction between messengers if something goes wrong, sorry.

closed 2022-08-04 by mdevaev

JokerGermany, 2022-08-04:
Okay thought you would do care about free software.

Now i know better, thanks.
https://www.youtube.com/watch?v=uvNkdAggUGU

mdevaev, 2022-08-04:
I try to use ready-made services, and since I am the only one who admins all public channels, my support capabilities are limited. Maybe in the future I will reconsider this decision regarding Matrix.

chri2, 2023-04-04:
> You don't even need a phone number

I'd like to open this again: I just tried to get through with a simple question I thought not worth an issue, but couldn't log into discord. The page said I'd need to provide a phone number to protect the fellow discord users from such dangerous individuals like I might be (the one with a firefox browser configured not to show much data to collect and to identify).

See the screenshot. The important part is "Verifizieren über Telefon" - "verify using a phone". I'm supposed to trust them with a phone number?

[image]

mdevaev, 2023-04-04:
> I'm supposed to trust them with a phone number?

Yes. I understand that some users will not be happy about this (me too btw), but this is an inevitable evil. Your information has already been sold hundreds of times, including by your mobile operator. Discord won't make it any worse.

chri2, 2023-04-04:
_"Lower your shields and surrender your ships. We will add your biological and technological distinctiveness to our own. Your culture will adapt to service us. Resistance is futile."_ (The Borg)

Might be that somewhere someone already stores something about me - inevitably. But every little bit less helps.

I really understand that you'd like to rely on existing infrastructure and I would do the same.

There is other existing infrastructure that works and is probably already used by most of the users of your project. 

If you define GitHub as "inevitable evil" (which I wouldn't sign, because there are alternatives for this, too) you could offer [discussions](https://docs.github.com/en/discussions/quickstart) on GitHub as an alternative.

But still: I understand that this is all additional work and I just wanted to add to this issue that the "no phone number needed" is not true (anymore or for privacy oriented browsers).