pikvm/pikvm #797: [UPDATED] HowTo add OLED for v2 DIY

arch1mede, 2022-08-05:
This is mostly for v2 DIY and NOT intended for the v3. Do NOT follow this [howto.](https://github.com/pikvm/pikvm/issues/348)

This process is made easier if you install the v3 [image](https://files.pikvm.org/images/v3-hdmi-rpi4-box-latest.img.xz). (Reminder: v3 image on a v2 = yes, v2 image on a v3 = no)

OLED screen [wiring](https://pinout.xyz/)
VCC 3.3v is pin 1
SDA pin 3
SCL pin 5
GND pin 9

The following steps are NOT needed if using the v3 image as stated above

```
# In order to make this work on v2, follow the directions below:
# 1. Add "i2c-dev" without the quotes into /etc/modules-load.d/kvmd.conf file
# 2. Add "dtparam=i2c_arm=on" without the quotes into /boot/config.txt file
# 3. Update /usr/lib/systemd/system/kvmd-oled.service file
"""
Change this line FROM:

ExecStart=/usr/bin/kvmd-oled --height=32 --offset-y=6 --clear-on-exit

TO

ExecStart=/usr/bin/srepac-oled --height=32
"""
# 4. Enable kvmd-oled services via "systemctl enable --now kvmd-oled" (applies to both v2/v3)
# 5. reboot on v2 and watch the magic on your oled screen; no need to reboot v3
###
```

closed 2022-08-07 by mdevaev

wzqvip, 2023-10-05:
I use raspberrypi zero w(out dated) with a 12864 Oled display. Now I have followed the script and it functions like a 12832 device. I tried to use --height=64, but it still display two lines only and went back to 12832 when reboot. How can I change the resolution for pikvm. Thanks.

waza-ari, 2023-12-17:
Hey there, I know that's a bit of a necro, but as that's the first thing popping up on google and the question there is the same I had, let me quickly comment. The usual [luma flags](https://github.com/rm-hull/luma.examples/blob/master/README.rst) (see here for reference, haven't found another place where they're documented) work just fine.

They have to be set in the systemd service file (usually `/usr/lib/systemd/system/kvmd-oled.service` when using the image), and keep in mind they need to be added twice (once for the logo splash screen, once for the actual display). In my case I'm using a different device driver and height 64:

```
ExecStartPre=/usr/bin/kvmd-oled --display sh1106 --height 64 --interval=3 --clear-on-exit --image=/usr/share/kvmd-oled/hello.ppm
ExecStart=/usr/bin/kvmd-oled --display sh1106 --height 64
```

Then, reload the systemd daemon to pick up the changes and restart the service:

```
systemctl daemon-reload
systemctl restart kvmd-oled
```

This should be persistent across reboots. Just keep in mind its a RO filesystem by default and you need to issue `rw` before doing any changes.

MatteoRagni, 2025-01-10:
Hello there, I tried this with the image for DIY PiKVM V2 Platform (in my specific case the [HDMI-CSI version](https://files.pikvm.org/images/v2-hdmi-rpi4-latest.img.xz)). I'm adding info about my process, hoping that it may be helpful for other users, since this thread was extremely helpful for me (thanks!).

My display is an old Adafruit SSD1306 128x64 (non-STEMMA version). In my specific case I had to close the jumpers SJ1 and SJ2 to enable i2c, as reported in the [instructions](https://learn.adafruit.com/monochrome-oled-breakouts/wiring-128x64-oleds).

[image]

The pinout are:

| Display pin name | Display pin number | PiKVM (RPi4) pin name | PiKVM (RPi4) pin number |
|------------------|--------------------|-----------------------|-------------------------|
| GND | 8 | GND | 9 |
| 3v3 | 6 | 3v3 Power | 1 |
| Data | 1 | GPIO2 (Data) | 3 |
| Clk | 2 | GPIO3 (Clock) | 7 |

To make this specific display working on the image I had to install the `i2c-tools` package (**after** the operations described in [initial comment](https://github.com/pikvm/pikvm/issues/797#issue-1330088148)).

```
rw && pacman -Sy i2c-tools
```

The package inastall an utility that allows to inspect for i2c devices connected:

```
i2cdetect -y 1
```

For my specific display, the i2c address is 0x3D, which differs from the luma default (that is 0x3C).

I had to edit the `/usr/lib/systemd/system/kvmd-oled.service` setting the following flags:

```
ExecStartPre=/usr/bin/kvmd-oled --display ssd1306 --height 64 --i2c-address 0x3D --interval=3 --clear-on-exit --image=/usr/share/kvmd-oled/hello.ppm
ExecStart=/usr/bin/kvmd-oled --display ssd1306 --height 64 --i2c-address 0x3D
```

and enabling / starting the daemon as described in [this comment](https://github.com/pikvm/pikvm/issues/797#issuecomment-1859249320).