pikvm/pikvm #799: Building v0-hdmiusb build Error 2

RobCod, 2022-08-06:
**Describe the bug**
make os fails to build the os after Updating the trust database

**To Reproduce**
Steps to reproduce the behavior, like:
1. build os
2. build os refuses to run
3. sudo build os, downloads everything starts to build and then fails after Updating trust database (step 26/90)

**Expected behavior**
To build sucessfully so i can move onto the make install portion

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: arch 64bit
 - Browser: Firefox
 - Version 103?

**PiKVM info:**
  - Raspberry Pi board version: pi 3 B+
  - PiKVM platform: v0-hdmiusb
  - Video capture type: ezcoo 4 switch KVM hub
  - KVMD version: `pacman -Q | grep kvmd` - Returns with nothing
  - uStreamer version: `pacman -Q | grep ustreamer` - Also returns with nothing
  - Linux kernel: `uname -a` - Linux Desktop 5.18.16-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 03 Aug 2022 11:25:04 +0000 x86_64 GNU/Linux

mdevaev, 2022-08-06:
Retry again

RobCod, 2022-08-06:
I have several times

RobCod, 2022-08-06:
I recloned the github repo, and have the same issue on ubuntu

b0d8mr4zu, 2022-08-06:
The latest build of the base image is probably broken.
Fast workaround:
Change `REPO_URL ?= http://de3.mirror.archlinuxarm.org` in Makefile to `REPO_URL ?= http://tardis.tiny-vps.com/aarm/repos/2022/08/01` to use archived image. Or add it to the config.mk.
The repo contains images archived from http://mirror.yandex.ru/archlinux-arm. More info in https://archlinuxarm.org/forum/viewtopic.php?f=30&t=9542 .
This is not a fix though.

mdevaev, 2022-08-16:
At the moment, the build is successful. Probably the base image has already been updated. Run `rm -rf .pi-builder` and `make os NC=1` to rebuild with flushed caches.