pikvm/pikvm #801: iPad Pro 11" (3rd Gen) screen flicker

fxbulldog, 2022-08-08:
PiKVM v3 Pre-Assemble
ezcoo HDMI 2.0 KVM Switch 4x1
Enabled Dual mouse mode - Had to select Relative to get the mouse to work.

Issue:
Screen flickers to the point that it is not usable.  Flicker = screen is seen then not seen every 10 seconds
Both video (MJPEG/HTTP & H.264 / WebRTC) modes tied 

Try 1 (Using Apple USB-C to Digital AV):
Connected to KVM switch select GPIO input (Screen Flicker)
Rebooted PiKVM and Switch (Screen Flicker)
Connected direct to PiKVM as a single source (Screen Flicker)

Try2 (Using Akhara USB-C to Digital AV):
Connected direct to PiKVM as a single source (Screen Flicker)

Try3 (Using Dell Thunderbolt Dock WD19TBS):
Connected direct to PiKVM as a single source (Screen Flicker)

fxbulldog, 2022-08-08:
Updated to the following it is better in the MJPEG/HTTP video mode in that is does not go to no signal and just fades for 1sec instead. 

KVMD: 3.134
Streamer: 5.19 (ustreamer)
HAS_PDEATHSIG: Yes
WITH_GPIO: Yes
WITH_PTHREAD_NP: Yes
WITH_SETPROCTITLE: Yes
WITH_SYSTEMD: Yes
Linux kernel:
Machine: armv7l
Release: 5.15.34-4-rpi-ARCH
Version: #1 SMP Tue Jul 5 12:10:16 MSK 2022

arch1mede, 2022-08-11:
Hi, the tablet experience still needs to be worked on, the dev is aware but is not a priority at the moment. I have found that activating VNC+using a VNC client is a much better experience. Please see this [page.](https://docs.pikvm.org/vnc/)

My VNC client of choice is Jump but its a little pricey.

mdevaev, 2022-10-06:
Could you show `journalctl -u kvmd` on flickering?

mdevaev, 2022-11-10:
I've tried iPad Pro 12.9 with the official Apple USB->HDMI converter and everything is working great.

+ no answer from the issue author, so I'm closing this.

closed 2022-11-10 by mdevaev