pikvm/pikvm #803: Ability to use LEDs in GPIO title

ivareri, 2022-08-11:
**Is your feature request related to a problem? Please describe.**
I want a way to see GPIO input states at a glance. 

My specific use case is to check which channel is active on a GPIO controlled KVM. 

[image] 

**Describe the solution you'd like**
An easy way to include LEDs in GPIO title

**Describe alternatives you've considered**
Including HTML in the yaml config

**Additional context**
This config does what I want, but it's a bit messy
```
kvmd:
    gpio:
        scheme:
            PC1:
                mode: input
                pin: 17
            PC2:
                mode: input
                pin: 27
            button1:
                mode: output
                pin: 22
                switch: false
        view:
            header:
                title: 'KVM PC1 <img id="gpio-led-PC1" class="gpio-led inline-lamp-big led-gray" src="/share/svg/led-circle.svg" data-color="green"> PC2 <img id="gpio-led-PC2" class="gpio-led inline-lamp-big led-green" src="/share/svg/led-circle.svg" data-color="green">'
            table:
                - ["#KVM Input selection"]
                - []
                - ["button1|Change"]
```

mdevaev, 2022-08-12:
Got it. I'll make it but later.

mdevaev, 2022-10-08:
It's done & released. You can use it like this:
```yaml
title: ["#KVM", "#PC1", "PC1", "#PC2", "PC2"]
```

closed 2022-10-08 by mdevaev