pikvm/pikvm #804: Add crontab support for macro script invocation

alexoms, 2022-08-11:
The current macro event timing is limited to a Delay wait event of n milliseconds.  There is however a need to trigger macro scripts based on a certain time of day or even day of week (eg, weekdays vs weekends) or recurring pattern. 

**Describe the solution you'd like**
 One might envision a crontab formatted feature to be added such that any number of scripts can be run against a cron.  

**Describe alternatives you've considered**
Creating another macro event that is different from the delay event is probably shortsighted.  But a crontab formatted entry which would be associated a script would then enable all sorts of automation!

**Additional context**
Crontab reference formats: https://alvinalexander.com/linux-unix/format-of-crontab-date-time-fields-examples/

Sample format:

# run the macro script named login-and-move-mouse.json at 4:30am
30 4 * * * ~/piKVM/macros/login-and-move-mouse.json

arch1mede, 2022-08-15:
You can install cronie
pacman -Ss cron
pacman -S cronie

arch1mede, 2022-09-14:
There is cronie (need to install) or you can use the systemd timer but having this integrated into pikvm would be a feature request for the dev. Also, for security reasons, macros's are never saved on the pikvm, this behavior would also need to be changed.