pikvm/pikvm #809: Either way to create .img file for PIKVM

zalouma, 2022-08-29:
Hey Guys

Would like to thank you for such great project, myself been using pikvm for many projects however my only issue was always mounting anything bigger than 2.2GB to get this going and to create .img was always pain in the neck however i found a very easy way and though to share it to this create community without all the steps here https://docs.pikvm.org/msd/#upload-images-manually-without-web-ui

I would start buy getting a USB flash drive, install ventoy on it over windows and then move all the iso files i want over the ventoy empty drive, my current setup has many linux distro iso and windows iso then once all looks good on that iso

I would install this application which is the game changer https://www.alexpage.de/download/usbit/usbit.zip , then run it and choose that usb and it will create the .img file for me

Then I follow these steps to upload it to pikvm

Step 1: Remount internal storage to rw (read-write):

# kvmd-helper-otgmsd-remount rw

Step 2: Upload the .img image(s) to /var/lib/kvmd/msd/images via scp or similar.

Step 3: Create an empty file in /var/lib/kvmd/msd/meta/ with the exact name (case sensitive!) of the uploaded image. This will indicate PiKVM that the uploaded image is okay and can be used. For example:

/var/lib/kvmd/msd/meta/ubuntu-18.04.4-desktop-amd64.iso.complete

Step 4: Remount internal storage back to ro (read-only):

# kvmd-helper-otgmsd-remount ro

Once all this is done, you have img that contain several iso setup on your pikvm ready for deployment anytime, then once i have this done i took a backup of the sdcard image for any future needs

Though to give back to the community to keep the chain going and thanks again, sorry if i posted it under new issues, couldnt find any other option to share this here

Best,
Zal

spoelstraethan, 2023-08-05:
I think the problem with this approach is you are basically doing the same thing that `dd if=/dev/sda1 of=/path/to/virtual/media` does, and you have to prepare all the ISOs you want on the Ventoy USB before capturing, and if you update even a single file you likely have to recreate the whole thing.

Or you could learn to use `losetup` to mount an image as a folder and drop in what you need, so you don't need an extra several layers of indirection that make it impossible for anybody to help you troubleshoot why it isn't booting.

zalouma, 2023-08-06:
@spoelstraethan well I know that already, but how is that what i need? Why would i have to worry about keep adding and removing while i can do it one time and be over with it and have everything i need ;) thank you for the suggestion though