pikvm/pikvm #810: zerow iso no kvmd-edidconf command

wnark, 2022-08-29:
**Describe the bug**
zerow iso no kvmd-edidconf command

**To Reproduce**
Steps to reproduce the behavior, like:
1.flash iso:Raspberry Pi Zero W (legacy), v2 platform
[image]
2.use kvmd-edidconf command
3.show this
[image]

**Expected behavior**
I can edit edid using kvmd-edidconf

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser chrome
 - xshell

**PiKVM info:**
  - Raspberry Pi board version : zero w
  - PiKVM platform: v2-hdmi
  - Video capture type：csi-bridge
  - KVMD version:kvmd 3.48-1
  - uStreamer version:ustreamer 4.9-1
  - Linux kernel: Linux pikvm 5.10.52-13-ARCH #1 SMP Sat Nov 20 06:29:40 MSK 2021 armv6l GNU/Linux

mdevaev, 2022-08-29:
ZeroW is not supported anymore.

closed 2022-08-29 by mdevaev

wnark, 2022-08-29:
https://github.com/pikvm/pikvm/blob/ddd1baf604bf9631eb1938862f6ed125973dea15/docs/community_faq.md
Found the old way of setting