pikvm/pikvm #812: The certificate has expired

wnark, 2022-08-30:
**Describe the bug**
When using wget to download, it prompts: The certificate has expired

**To Reproduce**
1. Download the image from the official website, and then swipe into the sd card:v2-hdmi-rpi4-latest.img.xz
2. pacman -Syu
3. wget https://raw.githubusercontent.com/th3drk0ne/DKN-Fan-Controller/main/install-pikvm.sh
4. show :The certificate has expired

**Expected behavior**
can be downloaded normally

**Screenshots**
[image]

**PiKVM info:**
  - Raspberry Pi board version ：RPi 4
  - PiKVM platform v2-hdmi
  - Video capture type CSI bridge
  - KVMD version: kvmd 3.141-1
  - uStreamer version: ustreamer 5.21-1
  - Linux kernel: Linux pikvm 5.15.34-4-rpi-ARCH #1 SMP Tue Jul 5 12:10:16 MSK 2022 armv7l GNU/Linux

mdevaev, 2022-09-02:
You have a problem with system clock. Check `timedatectl`

wnark, 2022-09-07:
> You have a problem with system clock. Check `timedatectl`

Hello, I checked that the time is correct, the only possibility is that the time zone is wrong, so I modified the time zone, the local time is completely correct after modifying the time zone, but I still found the same error
[image]

wnark, 2022-09-07:
> You have a problem with system clock. Check `timedatectl`

It may be a bug of wget
https://archlinuxarm.org/forum/viewtopic.php?f=57&t=16070

mdevaev, 2022-09-07:
You can use curl instead of wget in this case.

mdevaev, 2022-09-07:
In any case, this problem is not on our side.

closed 2022-09-07 by mdevaev