pikvm/pikvm #813: Web UI: Please allow prompt before closing page to be optionally disabled

dpb1, 2022-08-31:
**Is your feature request related to a problem? Please describe.**

I often pop onto my pikvm to check something and close the window when finished.  It's jarring to me to have to confirm I want to close it.  My reasoning behind why this does not fit my mental model is I can simply reconnect to the pikvm and have it continue to be open where it was.  This is exactly why I have a pikvm, so I can have a persistent connection to the box KVM when I need it.  The pikvm is sitting there ready for me to reconnect with a simple ctrl-alt-t if I did this accidentally.

**Describe the solution you'd like**

I'd like for this to be an option in the webui with a slider to disable it.   This would probably fit at the bottom of the 'System' drop down.  Defaulting to on would be fine, and less jarring for people who are used to the current mode of operation.

```
   Warn before closing page [off / on]
```

Alternatively, if there were some other way to adjust settings, it would be fine to be there as well.  I'm not an expert in what ways to adjust options exist.

**Describe alternatives you've considered**

I have looked at installing custom javascript for pikvm in my browser through tampermonkey / etc to disable the onpageunload() or whatever bindings that are used, but I do not like maintaining this kind of script in my browser, as I have to do it across systems, and it introduces another thing into my system that gives a chance for misconfiguration and attackers.

This kind of workaround is described here: https://superuser.com/questions/705307/how-can-i-disable-are-you-sure-you-want-to-leave-this-page-popups-in-chrome

I have also asked in the support channel on discord where I received this information.

**Additional context**

Prompt when closing (in chrome):
[image]

Notice also that the text in the box (which might be the stock text chrome gives when a page is changed) makes it seem as though I have changed something that needs to be saved before I leave this page.  Which is also jarring.  I spent a few minutes digging through menus to figure out what I could be changing to make this prompt show up.  Eventually I realized I was not changing anything, instead this was the default behavior of pikvm.

Thanks for your consideration, and thank you for the **fantastic** open source project.  Truly amazing given the commercial alternatives.

mdevaev, 2022-09-01:
Hello! I've just added this:
[image]
I hope this solves your problem.

mdevaev, 2022-09-01:
Released.

closed 2022-09-01 by mdevaev