pikvm/pikvm #817: Problem with PiKVM and pacman for Tailscale Install

nmitchell86, 2022-09-12:
Hi, and thank you for looking at my problem here with the PiKVM. Overall, I'm just trying to install tailscale so that I can access my PiKVM and the attached computer from the internet. I'm not familiar with Linux but I can follow directions pretty well. The issue I'm having is getting the first command to complete. Please see all the info below.

**Describe the bug**

I'm trying to install Tailscale but I'm getting an error. I run the command:
```
 rw
 pacman -Syu
```

I get an error:
```
:: Synchronizing package databases...
 core.db failed to download
error: failed retrieving file 'core.db' from de3.mirror.archlinuxarm.org :
error: failed to synchronize all databases (download library error)
```

I can't continue to install Tailscale. I read that updating the PiKVM software might fix things but that is not working either.

When running command: 
```
 rw
 pacman -Syu
```

I get the same error as before:
```
:: Synchronizing package databases...
 core.db failed to download
error: failed retrieving file 'core.db' from de3.mirror.archlinuxarm.org :
error: failed to synchronize all databases (download library error)
```

**PiKVM info:**
  - Raspberry Pi board version: Raspberry Pi 4
  - PiKVM platform: PiKVM v3 HAT
  - KVMD version: `pacman -Q | grep kvmd`
```
kvmd 3.143-1
kvmd-fan 0.22-1
kvmd-oled 0.13-1
kvmd-platform-v3-hdmi-rpi4 3.143-1
kvmd-webterm 0.42-1
```
  - uStreamer version: `pacman -Q | grep ustreamer`
`ustreamer 5.21-1`
  - Linux kernel: `uname -a`
`Linux pikvm 5.15.34-4-rpi-ARCH #1 SMP Tue Jul 5 12:10:16 MSK 2022 armv7l GNU/Linux`

**Additional context**
In a web browser I can get to `de3.mirror.archlinuxarm.org` but on that page I do not see a file called `core.db`. But I'm not a Linux user so I could be missing something on that part. I can say that it looks like all the items on the `de3.mirror.archlinuxarm.org` page was very recently updated. Could the command need to be updated because the file is in a different location or something? I'm not sure why the PiKVM can't get the file `core.db`. I'm not a Linux user let alone a Linux admin so I don't really understand `pacman -Syy` and `pacman -Syu` commands and how to modify them.

Any help that can be provided will be greatly appreciated. Thank you.

mdevaev, 2022-09-12:
Hello. Just try again a little later. This is a temporary error.

nmitchell86, 2022-09-12:
Hi @mdevaev, I ended up pulling the flash card and starting over. The commands work now that I have a new install of the OS image.

mdevaev, 2022-09-12:
Pretty weird but I'm glad that it's fixed.

closed 2022-09-12 by mdevaev

blwhpenguin, 2024-08-01:
I know this is pretty old but I ran into the same problem and simply had to update my os, not sure if this is even the same os but I'm pretty sure it is, worth a shot for anyone having this issue though.

mdevaev, 2024-08-01:
@blwhpenguin please join to discord for support: https://discord.gg/bpmXfz5