pikvm/pikvm #818: odroid-hc4 bios/boot corrupted picture

0xFEEDC0DE64, 2022-09-16:
**Describe the bug**
I am trying to debug an odroid-hc4 in my rack and I connected the pikvm to the odroid's hdmi output. After the kernel boots successfully from sdcard, i can see the login prompt on pikvm, but I cannot watch boot procedure before the prompt and also I do not see the bootloader's output. There is a "efi like" menu / shell before the kernel boots and I can control it with a usb keyboard, but I cannot see any picture on pikvm. The screenshot below is stuck during the whole boot procedure even though many pixel may change when I navigate blindly through the bios menu.

**To Reproduce**
Connect pikvm to a odroid-hc4 (even without a sdcard) and try to see the boot procedure.

**Expected behavior**
The screen is readable and the colors are correct / not distorted.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Chrome
 - Version 105.0.5195.102 (Official Build) (64-Bit)

**PiKVM info:**
  - Raspberry Pi board version RPi 4
  - PiKVM platform: the kickstarter metal enclosure, i think I had to assemble a separate pcb ontop of a pi to get it working
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`
  
 (I temporarily lost ssh access to my pikvm at home, will update once I'm back)

mdevaev, 2022-09-16:
Droid gives incorrect resolution information. Try to force 1280x720 on odroid.

0xFEEDC0DE64, 2022-09-16:
i don't believe I can do such things in the pre-boot environment, the bios thing looks very basic and is only there to boot the real kernel from the sd card, I am trying to debug a not booting odroid, is there any other way? My old samsung lcd monitor can show the image, how does it handle this incorrect informations?

mdevaev, 2022-09-16:
Your monitor can handle 1080p60Hz, but V3 supports maximum 50Hz. You can try alternative EDIDs: https://docs.pikvm.org/edid

mdevaev, 2022-09-23:
Sup?

mdevaev, 2022-09-25:
Solved I guess. Let me know if not.

closed 2022-09-25 by mdevaev