pikvm/pikvm #819: Mac keyboard code problem

visit1985, 2022-09-17:
First, a big **Thank you for your awesome work!** I know you just closed an annoying ticket like this, so I sincerely apologize for this one.

I access a Windows 10 Host with US keyboard layout via PiKVM from Mac OS 12.5 with an Englisch International Magic Keyboard and U.S. layout.

In Safari the &#96;~ key is correctly sent to the web interface.
In Chrome the &#96;~ key is interpreted as N/US.

I have to use Chrome in app mode to avoid the Ctrl+T problem, but I need the &#96;~ key pretty often too.
Is there a way to work around this?

mdevaev, 2022-09-18:
Hello. This one?

[image]

I'll try to debug this.

visit1985, 2022-09-18:
Yes, this one. If I can help you debug, just let me know how.

mdevaev, 2022-09-18:
Yes, I need your help here :)

You need to open the JS console in chrome and use this url for KVM: `https://yourkvm/kvm/?debug=1` Press that keys and show me the JS console log.

visit1985, 2022-09-18:
The &#96;~ key logs

```
Keyboard: key pressed: IntlBackslash
```

The \\| key produces the same chars, but logs

```
Keyboard: key released: Backslash
```

visit1985, 2022-09-18:
Safari logs IntlBackslash on §± key and Backquote on &#96;~ key. So it seems this keys need to be swapped.

mdevaev, 2022-09-18:
Okay, thank you. I'll think what I can do.

mdevaev, 2022-09-18:
Is it reproducet if you will use the regular pc keyboard with mac?

visit1985, 2022-09-18:
A regular pc keyboard does not have this key.

visit1985, 2022-09-18:
And I'm still curious, why it works properly in the web console, just not in KVM.

mdevaev, 2022-09-18:
The reason is that web console and KVM are using different ways to process keyboard.

mdevaev, 2022-09-20:
Here the some explanation of the issue: https://github.com/microsoft/node-native-keymap/issues/15

I can add menu switch to swap two keys but I need idea how to name this option shortly. `Swap IntlBackslash/Backqoute keys is too long.

visit1985, 2022-09-20:
You could just make it a config switch in override.yaml

mdevaev, 2022-09-20:
This is a browser problem and should be solved in the browser, in override the option would work for everyone

visit1985, 2022-09-20:
I did some further testing and found, that on a javascript keydown event the event.key attributes is always set correctly, while the event.code attribute is different between Chrome and Safari. Couldn't you implement a condition based on the detected browser, event.key and event.code then?

mdevaev, 2022-09-20:
What happens if you use a regular pc keyboard on mac os? I checked on my mac and a similar fix for apple magic keyboard breaks pc keyboard because there is no way to distinguish different keyboards at the browser level.

visit1985, 2022-09-20:
The only "regular keyboard" I own is [this](https://www.amazon.com/dp/B01DBVH16U). At least it has a &#96;~ key next to 1.
The key/code combinations to fix do not appear with that keyboard on my mac.

| Browser | Key pressed  | event.key | event.code | event.shift |
| --- | --- | --- | --- | --- |
| Chrome | `` `~ `` | ` | Backquote | false |
| Chrome | `Shift` + `` `~ `` | ~ | Backquote | true |
| Safari | `` `~ `` | ` | Backquote | false |
| Safari | `Shift` + `` `~ `` | ~ | Backquote | true |

mdevaev, 2022-09-20:
If I just swap keys `Backquote` and `IntlBackslash` for non-Safari, your PC keyboard will stop working correctly and will generate IntlBackslash when you click on `~`. This setting should depend on the type of keyboard, but there is no way to know which keyboard you are currently using.

visit1985, 2022-09-21:
Do you have access to event.key/code from your kvm code?

You could build something like:

if (event.key = "`" and event.code = "IntlBackslash") then event.code = "Backquote"

mdevaev, 2022-09-21:
Good idea. I'll try

visit1985, 2022-09-21:
Looks like `x in []` has to be replaced with `[].includes(x)` in javascript.

```
if (code == "IntlBackslash" && ["`", "~"].includes(event.key)) {
	code = "Backquote";
} else if (code == "Backquote" && ["§", "±"].includes(event.key)) {
	code = "IntlBackslash";
}
```

mdevaev, 2022-09-21:
Oh crap. One moment.

mdevaev, 2022-09-21:
3.147 released

visit1985, 2022-09-21:
Now it works perfectly on magic and external pc keyboard. Thanks a lot!

mdevaev, 2022-09-21:
Great! Thank you for your help.

closed 2022-09-21 by mdevaev