pikvm/pikvm #820: iOS VNC does not pass capitol letters to the server

arch1mede, 2022-09-18:
Using JUMP or bVNC, does not pass capitol letters through to the server.

Pressing shift 1 time or double tapping shift doesn't do anything.

To reproduce:
iOS 16, iOS 15.7 does the same thing
Updated PiKVM
Reboot
Any lowercase letters are passed correctly

ashao2devops, 2022-09-18:
I also have the same issue, the Jump VNC to PiKVM, remote desktop is Windows 10, opened a notepad, type in via Jump virtual keypad, lower letters are good (show in notepad), capital letter not working, show a space in notepad, also a few other character also not working.

iPhone: IOS 16 or 15.x
Jump: version 9.6.8 build 90608 latest so far
PiKVM: latest version
Remote desktop: Windows 10 US-en keyboard

arch1mede, 2022-09-20:
UPDATE: A user has found a work around at least with JUMP, you can use the alternate SHIFT key on the bar instead.