pikvm/pikvm #821: Streamed resolution doesn't update when server's resolution is reduced

coderthetyler, 2022-09-21:


coderthetyler, 2022-09-21:
Learn something new every day. Root cause of this issue is that:
- Increasing the display resolution increases the active signal resolution, and
- Decreasing the display resolution **does not** always decrease the active signal resolution.

You can see this mismatch in the Advanced Display Settings:
[image]

The value for 'Active signal resolution' is what the PiKVM picks up. To change it, you just the display adapter options > list all modes > pick whatever resolution you want. This fixed the problem for me & the streamed resolution now matches what the desktop resolution is.

Thanks for the help!

closed 2022-09-21 by coderthetyler

mdevaev, 2022-09-21:
:ok_hand: