pikvm/pikvm #822: RFC: add more protocol bits in serial/spi mcu hid emulator to handle atx function as well.

tao-j, 2022-09-22:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to purchase an overpriced and under powered Pi with Broadcom chip to do this. In addition, there are many old computing devices lying around can be repurposed to be used as a kvm server. For video capture, there are many solutions other than CSI-HDMI bridge in USB/PCIE form.
However, there are limited options to be able to emulate an usb hid device as well as do some GPIO job. Luckily, kvmd already provided an example using cheap AVR/STM32 MCU. If we are able to extend it to support GPIO to handle power/reset. Then we complete the solution of using any linux device as a kvm server (like your home router running openwrt). Since most computing devices can receive a USB to serial adapter + USB hdmi capture. And this solution will have a cost around $15-$25.

**Describe the solution you'd like**
In kvmd/hid/ there are avr/stm32 implementations of serial to hid conversion. And there is a 'proto.h' that defines how this conversion is done. It is possible to move the opto-couplers to MCU side. And extend the protocol to exchange info on that part.

Currently, the serial/spi protocol transmits 8 bytes each time, and only 4 bytes are used. We can make use of the additional bytes. If not enough, we can send more bytes each time.
**I can contribute patches below** to provide this kind of improvement, given the owner of this project is open to:

- [ ] merge new mcu code with extra protocol bits into kvmd repo
- [ ] merge documentation update with instructions on how to deploy such reduced cost solution
- [ ] merge build scripts for deb/rpm in to kvmd repo

**Describe alternatives you've considered**
Wait for someone to do the same.

**Additional context**
If this feature is ever implemented and adopted, it may have business implication on pikvm.org.
Also it conflicts with pi in pikvm since it no longer requires pi.

mdevaev, 2022-09-25:
Hello. This sounds reasonable, but involves a number of technical difficulties. For example, we will need a separate protocol and driver so as not to break compatibility with existing installations. Another difficulty will be the refinement of KVM, since HID and ATX are completely unrelated subsystems. Right now we are working on a global modification of the HID firmware to ensure compatibility with STM32. After its completion, I will be ready for this task.

tao-j, 2022-09-25:
Thanks for your openness. 
Here is the todo list on my mind:
* port to one of esp32/rp2040/blackpill
* extend protocol
* write a seperate gpio driver talks to mcu using another serial port to test ATX/gpio
* if the above works well, make changes to the current serial driver to handle gpio as well

Each item take me 2-3 weeks to prepare the patch. I will send PR for review when ready.

mdevaev, 2022-09-25:
I would advise you not to do a clean version, because I'm very picky about architectural changes (because I will have to support it all lol). Write it, I'll do a review, maybe I'll rewrite something myself.

tao-j, 2022-10-31:
Quick update: 
+ A hacky port to esp32/rp2040 is finished. Currently, I am using a blackpill to debug the rp2040. So rp2040 will have priority fron now on.

+ A fedora copr has been built https://copr.fedorainfracloud.org/coprs/tao-j/pikvm/
It is derived from other specs found online (check github  for .spec). Right now it builds and installs, to use it first gencerts and then use systemd to start service. However, kvmd is not running properly due to reason below.
+ Debian packages exists for all dependency except kvmd. But there is already someone sharing that .dsc .

The problem now is that `kvmd` :
1.  It has many hard corded strings calling utilities that exist only on rpi platform. 
2.  Some gpio feature/dependency is kind of hard to remove. 

Still investigating on this issue. Took longer than I expected

tao-j, 2022-11-06:
> This sounds reasonable, but involves a number of technical difficulties. 

Indeed, inside the `kvmd` too many things are coupled/independent. Too much time is needed to make that possible.

So change of plan: 

we can have a USB to gpio (vendor class) which exposes `/dev/gpiochip*` so the atx/gpio subsystem does not require any change. See https://github.com/notro/pico-usb-io-board/wiki. And most linux distros have `dln2` module enabled. I am working on making that into a composite device (vendor class + serial) so that it serves as an usb to serial adapter as well.

Packaging of janus/ustreamer has been submitted to Fedora for review.
janus/ustreamer exists in Debian

Before submit kvmd to Fedora/Debian, `HwInfoSubmanager` has better to be implemented in a rpi agnostic way (with device tree and opt/vc/bin/vcgencmd). Maybe I will write a HwInfo calling '/sys'
```python
# "hw": HwInfoSubmanager(**config.kvmd.info.hw._unpack()),
```

mdevaev, 2022-11-06:
As a first approximation, I would recommend using a script that simulates vcgencmd, it would be the easiest.

Blue-Beaker, 2022-11-24:
HID/USB and GPIO implementation should be made into plugins in independent version so users can choose which one to use, or write their own plugins for their hardware.
For example, OrangePi Zero supports both sysfs-GPIO and USB emulation, so users can write plugins for them and use its builtin GPIO and USB for controls, without the need to use extra devices.

tao-j, 2022-11-24:
USB to UART+GPIO+I2C+SPI should be working on rpi pico using this PR: https://github.com/notro/pico-usb-io-board/pull/2

Since tinyUSB does not provide absolute descriptor https://github.com/hathach/tinyusb/pull/1363  (I can manually add it, but too much effort for now). My port of the rp2040/esp32/blackpill/bluepill hid can not be completed now. For someone who is interested in this, please take a look at the latest two commits here: https://github.com/freekvm/kvmd.

tao-j, 2022-11-24:
> As a first approximation, I would recommend using a script that simulates vcgencmd, it would be the easiest.

Thanks for the suggestion. But it requires lots of effort to study a binary.

mdevaev, 2022-11-24:
> > As a first approximation, I would recommend using a script that simulates vcgencmd, it would be the easiest.
> 
> Thanks for the suggestion. But it requires lots of effort to study a binary.

Not at all: https://github.com/pikvm/kvmd/blob/master/testenv/fakes/vcgencmd

I'm using this to run kvmd on my x86_64 laptop.

neutronstriker, 2023-04-26:
> Quick update:
>
> * A hacky port to esp32/rp2040 is finished. Currently, I am using a blackpill to debug the rp2040. So rp2040 will have priority fron now on.
> * A fedora copr has been built [...]

Maybe it would be helpful to have rp2040 acting as USB-UART/SPI bridge+GPIO that connects to another RP2040 which acts as HID emulator and if possible both firmware should determine the device role based on a GPIO state during Power-on. What do you think?

tao-j, 2023-04-29:
@neutronstriker the bridge you mentioned is here https://github.com/notro/pico-usb-io-board

As soon as https://github.com/hathach/tinyusb/pull/1363 gets merged, rp2040 port of the hid emulator can be tested and completed.

Auto switching can work, but usually we are setting those mcu up and leave them their untouched, so flashing two firmware to two borads should be find.

neutronstriker, 2023-04-29:
I agree since the setup is a one-time job, it's fine to have 2 different firmware pieces for rp2040's on each side of the connection.

mdevaev, 2023-08-10:
I'll add my five cents here.
* Firstly, I am in the process of deprecating the Arduino HID, it will be replaced by the [Pico HID](https://docs.pikvm.org/pico_hid/). It can also work with x86, if you use USB-TTL to connect to the UART of Pico.
* I really like the https://github.com/notro/pico-usb-io-board project, and since this works with regular GPIO drivers, no additional support is required from PiKVM.

So it remains just to buy a couple of Picos, a few cheap parts and everything will work. Considering what is written in the update to this issue, it can be closed. Individual patches are still being accepted, but given the cheapness of Pico, I see no reason to combine the GPIO module with HID.

closed 2023-08-10 by mdevaev