pikvm/pikvm #825: Shift key sticks

alexnagelberg, 2022-09-29:
**Describe the bug**
I saw a similar bug that describes EXACTLY what I see, but I don't believe there's any issue on my machine (#352). It seems to work okay in tigervnc, but in Jump any time I use shift it tends to stick. I've tried the latest beta releases to no avail.

Additionally, I have trouble sending combinations like ctrl + shift + <key> (specific case: gnome terminal in a virtualbox session over jump)

**To Reproduce**
1. Setup PiKVM with default settings and vnc settings
2. Connect to VNC using Jump Desktop (haven't tried many VNC clients as many don't work)
3. Type stuff and hold shift to modify a character
4. Watch it continue repeating character until you press another key

**Expected behavior**
It should press just the one key and send a key_up or whatever event needed to release

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 12
 - VNC client: Jump Desktop (latest stable/beta)

**PiKVM info:**
  - Raspberry Pi board version: Pi 4b
  - PiKVM platform: v3
  - Video capture type: HAT
  - KVMD version: 3.48-1
  - uStreamer version: `4.9-1`
  - Linux kernel: `5.10.52-13-ARCH`

mdevaev, 2022-09-29:
Could you write a video? Preferably with your fingers, so that I can see exactly how you press everything.

mdevaev, 2022-09-29:
Please perform a full update using `pacman -Syu` and try the latest software.

alexnagelberg, 2022-09-29:
I'll try the update and if it doesn't work, I'll try to put together a video of me typing... somehow...

mdevaev, 2022-10-06:
Perhaps it was fixed in the update.

closed 2022-10-06 by mdevaev

alexnagelberg, 2022-10-06:
Yep sorry just wanted to give it enough time to be sure but updating resolved it. Thanks!

mdevaev, 2022-10-06:
:ok_hand: