pikvm/pikvm #827: Pi-KVM

n3gwg, 2022-10-04:
PiKVM users, developers, et alia:

It is my understanding that PiKVM can work with Pi Zero2W and a CSI-2 bridge. I am curious, has anyone tried the PiKVM software, the CSI-2 bridge and a Banana Pi of any type?

If one uses a Pi Zero2W are there any features missing from if it were used with a Raspberry Pi 4B?

Just curious!

Stuart

mdevaev, 2022-10-04:
Hello. This is not officially supported and you need to prepare the operating system yourself. For the PiKVM software, you will need to replace some config files (such as UDEV rules) and patch the linux kernel. If you are a developer or an experienced system administrator, you will not have any problems with this. In addition, we are open to patches. If you need help with this, please contact us via [Discord](https://discord.gg/bpmXfz5).

closed 2022-10-04 by mdevaev

melroy89, 2023-05-16:
I actually just received the Banana PI BPI-CM4 myself as well.. It should be pin compatible and using the correct CSI-2 connection.
I will try to get this working with PiKVM and BlikKVM.  Related: https://forum.banana-pi.org/t/banana-cm4-vs-rasp-cm4-boot-loader/15494

Just Ubuntu Mate test: https://www.youtube.com/watch?v=wU6zKKlJtbQ