pikvm/pikvm #828: Any chance we can get this project ported to a SNC we can actually OBTAIN?

DigIntelligence, 2022-10-07:
Curious if we could get OFFICIAL PiKVM support (fork) for installation on something other than the unobtainable RPi.  Maybe OrangePi, BananaPi, LePotato?

I've been following this project for a while (while trying to obtain a Pi) and I gotta believe there is significant motivation for this....  Yes, I am a developer, but I don't need ANOTHER project at the moment.  All I would like is a finished IPKVM for an SBC I can actually OBTAIN without having to personally go down some rabbit hole (porting code).  (No Offense)

Found this: https://github.com/xe5700/kvmd-armbian   -  but I'm skeptical it has enough "inertia" to NOT require going down a similar conversion process to actually be implementable on a specific piece of hardware.

IOW: I would like a PiKVM solution to help me work on OTHER projects.  Not to become the actual project itself...  ;-)

Thanks for a GREAT solution in any case!

mdevaev, 2022-10-07:
Hello! The problem with porting to other platforms is that each of them will require efforts to support and fixing the USB gadget and video capture kernel drivers. I do this regularly and it takes a lot of time. Raspberry is the only one who offers sufficient technical support and open communications for this. Also all SBCs I know are not reliable enough: the kernel may freeze, devices may overheat or fail for other reasons. I'm not ready to make an official port and sign that the OrangeKVM(r)(c)(tm) will be reliable. Raspberry Pi is used precisely for reasons of strong reliability.

In addition, this project exists thanks to the sales of our hardware for Raspberry Pi and donations from their users. I buy a lot of hardware to make everything work. If I declare the support of others, my SBC fleet should be expanded to dozens of devices.

Therefore, I always tell interested users: I'm not against supporting alternative SBCs, I will accept patches for this, I will even support it, but I will not do all the work myself. All sources are open, the build system is simple and can also be modified to build packages and images for other SBCs. I will not sign their reliability, because it was done by third-party people, but I don't mind that the necessary code and images are lying here.

And finally, the chip shortage will end and you will be able to buy cheap Raspberries again :) If you really want to use the port on Orange or others, you can contact our discord, some users are doing this. But as far as I know, they could not get CSI capture to work there. This is the reason I mentioned about the difficulties of support.

closed 2022-10-07 by mdevaev

DigIntelligence, 2022-10-07:
Thank you for taking the time to provide a detailed response.  I understand and will support your position completely.

I've been watching this project for quite a while now and I guess I'll keep watching a bit longer until PI's actually become available again (at somewhat less than cutthroat prices).  This is not projected to happen anytime soon though....

mdevaev, 2022-10-07:
Let me know if you want to make port or find like-minded people. I will provide all possible support :ok_hand:

seamusdemora, 2022-10-10:
FWIW: The current non-availability of RPi boards does not seem to be related to a "chip shortage", but rather due to the organization's prioritization of commercial customers over individuals/hobbyists. This is according to statements made by Mr. Eben Upton; [REF](https://raspberrypi.stackexchange.com/a/139023/83790). My only reason for mentioning this is that perhaps if a *commercial party* in the business of selling KVM switches were to inquire about availability, that *commercial party* might get priority??

mdevaev, 2022-10-10:
I do not know and cannot speak for Raspberry. But our partners and I provide retail sales of V3 for individual and commercial customers, so if you want to buy it, there will be no problems.

seamusdemora, 2022-10-10:
I know you can't - and neither can I - that's why I included a link to some references. I'm happy for you, and maybe @DigIntelligence will find this useful. The reasoning behind the prioritization of commercial customers is perhaps a bit more clear now.

DigIntelligence, 2022-10-10:
Big business wins again!

Kind of ironic for a device which was originally adopted and driven by individual experimenters and hackers...

Time to move on I think...  RPi business decisions have effectively removed itself from our community and I think it would do us some good to look elsewhere to establish and re-focus our momentum.

$0.02

mdevaev, 2022-10-10:
You can contact the https://pishop.us, I think they will be able to allocate you a quota of ready-made pikvms.

DigIntelligence, 2022-10-10:
They have been (and are still showing) "Notify Me" in place of the order button.

Is this the case that you need to specifically "ask them" rather than just order one from them?

DigIntelligence, 2022-10-10:
...It shows "Notify Me" on the main page and then "Out of Stock" on the order page...