pikvm/pikvm #829: Local mouse cursor not syncronized with remote mouse

connected201, 2022-10-08:
Hi, i tried today to connect pikvm on my NVR, and i get this problem. How i can fix it? thanks.

https://user-images.githubusercontent.com/20506433/194716641-2de3f7c0-368d-4c81-8a7c-d7a4dda2de0b.mp4
```
KVMD: 3.153
Streamer: 5.23 (ustreamer)

    HAS_PDEATHSIG: Yes
    WITH_GPIO: Yes
    WITH_PTHREAD_NP: Yes
    WITH_SETPROCTITLE: Yes
    WITH_SYSTEMD: Yes

Linux kernel:

    Machine: armv7l
    Release: 5.15.68-2-rpi-ARCH
    Version: #1 SMP Fri Oct 7

 06:33:56 MSK 2022
```

mdevaev, 2022-10-08:
Hello. We have already [encountered this](https://github.com/pikvm/pikvm/issues/450), and apparently this is a mouse driver error in NVR. It is probably possible to make some kind of workaround, but without a sample device it is difficult to understand what exactly needs to be fixed.

As a workaround, I can suggest you try the relative mouse mode: https://docs.pikvm.org/mouse

Please let me know if this works. Knowing the NVR model would also help.

connected201, 2022-10-08:
thanks, relative mouse work for me.
[image]

i can give access to my pikvm connected to my nvr, this will help you to solve this problem?

closed 2022-10-08 by connected201

mdevaev, 2022-10-08:
For an absolute mouse, this is probably a specific bug in the NTR driver. Apparently, the driver incorrectly considers the proportions of the screen. Perhaps this can be fixed with specific mouse settings, but if the relative one works for you, then let's leave it that way. I'll just attach your case to another issue, and if I get a similar device, I'll study it in more detail with my local tools. Thank you :)