pikvm/pikvm #830: Add Support for Redfish Virtual Media

samcat116, 2022-10-12:
**Is your feature request related to a problem? Please describe.**
To use PiKVM in more enterprise environments, its useful to have IPMI and Redfish standard support. While PiKVM already supports Redfish for power operations, it would be great to include support for the virtual media APIs.

**Describe the solution you'd like**
Using the Redfish virtual media APIs, an image would be mounted to the host over USB.

**Describe alternatives you've considered**
None, but it appears there's already an API for something similar.

**Additional context**

mdevaev, 2022-10-12:
I'm not sure if it's possible now, but it's a good idea.

wrkode, 2025-11-27:
I'm wondering if this is still being researched?
@mdevaev, do you have any comments? do you need some help? :)

wrkode, 2025-11-28:
@mdevaev,
I see RedFish implementation here https://github.com/pikvm/kvmd, but I don't see a CONTRIBUTE.md.
I might be asking a stupid question, but how do I go about building a local testing/running image?
again, apologies for the stupid question

mdevaev, 2025-11-28:
> I have something working, but I need to setup up a build environment for
> testing POST operations since I need real MSD env and hardware (which I
> have :) ):
> VirtualMedia must be exposed under Managers
>
> Added Managers collections :
>
> redfishtool -S Never -u admin -p admin -r localhost:8080 root
> {
>     "Id": "RootService",
>     "Managers": {
>     },
>     "Name": "Root Service",
>     "RedfishVersion": "1.6.0",
>     "Systems": {
>     }
> }
>
> and added Managers Instance BMC
>
> redfishtool -S Never -u admin -p admin -r localhost:8080 Managers
> {
>     "Members": [
>         {
>         }
>     ],
>     "Name": "Manager Collection"
> }
>
> BMC Instance exposes VirtualMedia:
>
> redfishtool -S Never -u admin -p admin -r localhost:8080 Managers -I BMC
> {
>     "Description": "PiKVM Baseboard Management Controller",
>     "Id": "BMC",
>     "ManagerType": "BMC",
>     "Name": "PiKVM Manager",
>     "VirtualMedia": {
>     }
> }
>
> Listing the devices of VirtualMedia:
>
> redfishtool -S Never -u admin -p admin -r localhost:8080 raw GET /redfish/v1/Managers/BMC/Virta
> {
>     "Members": [
>         {
>         }
>     ],
>     "Name": "Virtual Media Collection"
> }
>
> and finally the media operations:
>
> redfishtool -S Never -u admin -p admin -r localhost:8080 raw GET /redfish/v1/Managers/BMC/VirtualMedia/CD
> {
>     "Actions": {
>         "#VirtualMedia.EjectMedia": {
>             "target": "/redfish/v1/Managers/BMC/VirtualMedia/CD/Actions/VirtualMedia.EjectMedia"
>         },
>         "#VirtualMedia.InsertMedia": {
>                 "URI"
>             ],
>             "target": "/redfish/v1/Managers/BMC/VirtualMedia/CD/Actions/VirtualMedia.InsertMedia"
>         }
>     },
>     "ConnectedVia": null,
>     "Description": "Virtual CD/DVD Media",
>     "Id": "CD",
>     "Image": null,
>     "ImageName": null,
>     "Inserted": false,
>     "MediaTypes": [
>         "CD",
>         "DVD"
>     ],
>     "Name": "Virtual CD"
> }

You just need to execute "make run" from kvmd repo.

Best regards,
Max Devaev

wrkode, 2025-11-28:
Tried,
 but I guess I'm missing something:

```
...snipped output
		&& mv /etc/kvmd/nginx/ssl /src/testenv/.ssl/nginx \
		&& mv /etc/kvmd/vnc/ssl /src/testenv/.ssl/vnc \
	"
+ mkdir -p /etc/kvmd/nginx/ssl
+ rm -f '/etc/kvmd/nginx/ssl/*'
+ cd /etc/kvmd/nginx/ssl
+ openssl ecparam -out server.key -name prime256v1 -genkey
+ openssl req -new -x509 -sha256 -nodes -key server.key -out server.crt -days 3650 -subj /C=US/O=PiKVM/OU=PiKVM/CN=localhost
+ chown root:kvmd-nginx /etc/kvmd/nginx/ssl/server.crt /etc/kvmd/nginx/ssl/server.key
+ chmod 440 /etc/kvmd/nginx/ssl/server.key
+ chmod 444 /etc/kvmd/nginx/ssl/server.crt
+ chmod 755 /etc/kvmd/nginx/ssl
+ mkdir -p /etc/kvmd/vnc/ssl
+ rm -f '/etc/kvmd/vnc/ssl/*'
+ cd /etc/kvmd/vnc/ssl
+ openssl ecparam -out server.key -name prime256v1 -genkey
+ openssl req -new -x509 -sha256 -nodes -key server.key -out server.crt -days 3650 -subj /C=US/O=PiKVM/OU=PiKVM/CN=localhost
+ chown root:kvmd-vnc /etc/kvmd/vnc/ssl/server.crt /etc/kvmd/vnc/ssl/server.key
+ chmod 440 /etc/kvmd/vnc/ssl/server.key
+ chmod 444 /etc/kvmd/vnc/ssl/server.crt
+ chmod 755 /etc/kvmd/vnc/ssl
sudo modprobe gpio_mockup gpio_mockup_ranges=0,40
modprobe: FATAL: Module gpio_mockup not found in directory /lib/modules/6.8.0-55-generic
make: *** [Makefile:110: /sys/module/gpio_mockup] Error 1
```

wrkode, 2025-11-29:
> Is the end of url `/redfish/v1/Managers/BMC/VirtualMedia/CD` depends on MediaTypes?

hum. @mdevaev ,
The device ID ("CD") in the URL `/redfish/v1/Managers/BMC/VirtualMedia/CD` is a fixed device identifier and is independent of the MediaTypes property.
I’ve implemented:

- The device ID is hardcoded as "CD".
- The MediaTypes property is set to ["CD", "DVD"].

This means the URL path is always `/VirtualMedia/CD`, but the `MediaTypes` property informs clients that the device supports both `CD` and `DVD` formats.

are you envisioning something different?

by the way, I'm not getting a grip with that gpio module. is there any way I can go around it? or shall I just build a pikvm OS version to test the work?

mdevaev, 2025-11-29:
A quick way to start: remove everything related to gpio from the Makefile, and remove testenv/override.d/10-gpio.yaml too. Please `20-atx.yaml` instead of it:

```yaml
kvmd:
    atx:
        type: disabled
```

After this `make run` should work.

mdevaev, 2025-11-30:
It is possible to stream images but I didn't write it yet: #101

We just need to decide how we use the local images. Maybe use a URL like `local:path/to/image.iso `? Or is it just that everything that doesn't start with http/https is considered local images?

mdevaev, 2025-11-30:
@wrkode I found the commit: https://github.com/pikvm/kvmd/commit/82ed7fc0d431dcb9e97ae11c77a86d0d5502cac8

I've also read some documentation on Redfish, let's try my implementation. Just tell me what you think about the `local:` prefix.

mdevaev, 2025-12-04:
@wrkode It's released now with a new bunch of changes, could you test it?

mdevaev, 2026-01-08:
@wrkode sup. Did you have a chance to test it?

moondev, 2026-02-22:
My tests were successful using the following:

```
curl -i -X POST \
    -H "Content-Type: application/json" \
    -L -k -u admin:admin https://<pikvmip>/redfish/v1/Managers/BMC/VirtualMedia/MSD/Actions/VirtualMedia.InsertMedia \
    -d '{
        "Image": "netboot.xyz.iso",
        "Inserted": true
    }'

```
If `netboot.xyz.iso` has been uploaded already, it activates and mounts the MSD with it.

eject also works
```
curl -i -X POST \
    -H "Content-Type: application/json" \
    -L -k -u admin:admin https://<pikvmip>/redfish/v1/Managers/BMC/VirtualMedia/MSD/Actions/VirtualMedia.EjectMedia
```
Testing with a url to media that has not been uploaded 
```
curl -i -X POST \
    -H "Content-Type: application/json" \
    -L -k -u admin:admin https://<pikvmip>/redfish/v1/Managers/BMC/VirtualMedia/MSD/Actions/VirtualMedia.InsertMedia \
    -d '{
        "Image": "https://releases.ubuntu.com/noble/ubuntu-24.04.3-live-server-amd64.iso",
        "Inserted": true
    }'
```
returns
```
{
    "ok": false,
    "result": {
        "error": "MsdUnknownImageError",
        "error_msg": "The image is not found in the storage"
    }
}
```

Which I believe is expected.

Many thanks for redfish support and great to see it continue to improve. I've been using it with  great success controlling multiple machines attached to pikvm switch with Canonical MAAS.

```
curl -L -k -u admin:admin https://<pikvmip>/redfish/v1/Systems
{
    "@odata.id": "/redfish/v1/Systems",
    "@odata.type": "#ComputerSystemCollection.ComputerSystemCollection",
    "Members": [
        {
            "@odata.id": "/redfish/v1/Systems/0"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort0"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort1"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort2"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort3"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort4"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort5"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort6"
        },
        {
            "@odata.id": "/redfish/v1/Systems/SwitchPort7"
        }

```

mdevaev, 2026-10-01:
In the latest release this feature is available for the Redfish API.

This connects a remote image via HTTP. Currently supported url schemes: `http://`, `https://`, `smb://` and `sftp://`. To pass username and password you need to add `UserName` and `Password` paramethers to JSON body.
```
curl -i -X POST \
    -H "Content-Type: application/json" \
    -L -k -u admin:admin https://<pikvmip>/redfish/v1/Managers/BMC/VirtualMedia/MSD/Actions/VirtualMedia.InsertMedia \
    -d '{
        "Image": "https://releases.ubuntu.com/noble/ubuntu-24.04.3-live-server-amd64.iso",
        "Inserted": true
    }'
```

This ejects the media:
```
curl -i -X POST \
    -H "Content-Type: application/json" \
    -L -k -u admin:admin https://<pikvmip>/redfish/v1/Managers/BMC/VirtualMedia/MSD/Actions/VirtualMedia.EjectMedia
```

There is no Web UI interface to configure this right now, it's kinds experimental release.

mdevaev, 2026-10-03:
Added NBD support with `nbd://`. Accepted URLs like `nbd://server:port/exportname` like in [official doc](https://t.me/riscypeople/22770) but without arguments.

mdevaev, 2026-10-11:
I've documented it and consider that this feature is ready: https://docs.pikvm.org/api/#mass-storage-drive-information

closed 2026-10-11 by mdevaev