pikvm/pikvm #832: pikvm smb

caochengdong, 2022-10-16:
[kvmd-webterm@pikvm ~]$ systemctl restart smb.service
Failed to restart smb.service: Access denied
See system logs and 'systemctl status smb.service' for details.
[kvmd-webterm@pikvm ~]$ su
Password: 
[root@pikvm kvmd-webterm]# systemctl restart smb.service
Job for smb.service failed because the control process exited with error code.
See "systemctl status smb.service" and "journalctl -xeu smb.service" for details.
[root@pikvm kvmd-webterm]# rw
[root@pikvm kvmd-webterm]# systemctl restart smb.service
[root@pikvm kvmd-webterm]#

caochengdong, 2022-10-16:
不好意思 不会英语  我想在pikvm下装smb服务

mdevaev, 2022-10-16:
Hello. You need to use `su -` to get the root access. After that you will be able to run `systemctl` commands. Please use Discord for the support questions: https://discord.gg/bpmXfz5

closed 2022-10-16 by mdevaev

caochengdong, 2022-10-16:
I'm sorry, I can't speak English. Plus, I have trouble using this software for the first time. I will pay attention