pikvm/pikvm #840: GPIO Input not working

andreaconfa, 2022-11-01:
I all,
i'm testing pikvm on a Raspberry PI4B with an ezcoo 4hdmi switch.

I'm building a custom HAT PCC to handle my four server ATX control.

In my config ovverride i setted up and correctly tested the ezcoo switch and is working fine.
I also added some input / outputs to control and receive status of the servers ATX but i found that the input are not displayed. The led bouble stays always grey.

I'm testing it manually puytting the GPIO tu the 3.3v directly and i found that is not working. 
Here my configu override:

`kvmd:
  2     gpio:
  3         drivers:
  4             ez:
  5                 type: ezcoo
  6                 protocol: 2
  7                 device: /dev/ttyUSB0
  8         scheme:
  9             ch1_led:
 10                 driver: ez
 11                 pin: 0
 12                 mode: input
 13             ch2_led:
 14                 driver: ez
 15                 pin: 1
 16                 mode: input
 17             ch3_led:
 18                 driver: ez
 19                 pin: 2
 20                 mode: input
 21             ch4_led:
 22                 driver: ez
 23                 pin: 3
 24                 mode: input
 25             ch1_button:
 26                 driver: ez
 27                 pin: 0
 28                 mode: output
 29                 switch: false
 30             ch2_button:
 31                 driver: ez
 32                 pin: 1
 33                 mode: output
 34                 switch: false
 35             ch3_button:
 36                 driver: ez
 37                 pin: 2
 38                 mode: output
 39                 switch: false
 40             ch4_button:
 41                 driver: ez
 42                 pin: 3
 43                 mode: output
 44                 switch: false
 45             pwr_status_1:
 46                 pin: 26
 47                 mode: input
 48             pwr_status_2:
 49                 pin: 19
 50                 mode: input
 51             pwr_status_3:
 52                 pin: 20
 53                 mode: input
 54             pwr_status_4:
 55                 pin: 21
 56                 mode: input
 57             pwr_cmnd_1:
 58                 pin: 5
 59                 mode: output
 60                 switch: false
 61             pwr_cmnd_2:
 62                 pin: 6
 63                 mode: output
 64                 switch: false
 65             pwr_cmnd_3:
 66                 pin: 7
 67                 mode: output
 68                 switch: false
 69             pwr_cmnd_4:
 70                 pin: 25
 71                 mode: output
 72                 switch: false
 73             hdd_status_1:
 74                 pin: 7
 75                 mode: input
 76             hdd_status_2:
 77                 pin: 9
 78                 mode: input
 79             hdd_status_3:
 80                 pin: 10
 81                 mode: input
                hdd_status_4:
 83                 pin: 11
 84                 mode: input
 85             pwr_reset_1:
 86                 pin: 14
 87                 mode: output
 88                 switch: false
 89             pwr_reset_2:
 90                 pin: 15
 91                 mode: output
 92                 switch: false
 93             pwr_reset_3:
 94                 pin: 16
 95                 mode: output
 96                 switch: false
 97             pwr_reset_4:
 98                 pin: 17
 99                 mode: output
100                 switch: false
101         view:
102             header: 
103                 title: Comandi Server / KVM 
104             table:
105                 - ["#HDMI Switch"]
106                 - ["#Server 1", ch1_led, ch1_button]
107                 - ["#Server 2", ch2_led, ch2_button]
108                 - ["#Server 3", ch3_led, ch3_button]
109                 - ["#Server 4", ch4_led, ch4_button]
110                 - []
111                 - ["#Server 1"]
112                 - ["#Power / HDD", pwr_status_1. hdd_status_1]
113                 - ["#Power Command", pwr_cmnd_1]
114                 - ["#Reset Command", pwr_reset_1]
115                 - []
116                 - ["#Server 2"]
117                 - ["#Power / HDD", pwr_status_2, hdd_status_2]
118                 - ["#Power Command", pwr_cmnd_2]
119                 - ["#Reset Command", pwr_reset_2]
120                 - []
121                 - ["#Server 3"]
122                 - ["#Power / HDD", pwr_status_3, hdd_status_3]
123                 - ["#Power Command", pwr_cmnd_3]
124                 - ["#Reset Command", pwr_reset_3]
125                 - []
126                 - ["#Server 4"]
127                 - ["#Power / HDD", pwr_status_4, hdd_status_4]
128                 - ["#Power Command", pwr_cmnd_4]
129                 - ["#Reset Command", pwr_reset_4]
130`

mdevaev, 2022-11-03:
Hello. Are you sure that you're using corrent pin numbers? You need to use GPIO(X) values, not physical pin numbers: https://pinout.xyz

mdevaev, 2022-11-06:
Sup?

andreaconfa, 2022-11-07:
Issue resolved.
I’m the configuration I accidentally used a pin two times

Inviata da iPhone

mdevaev, 2022-11-07:
Glad to hear it!

closed 2022-11-07 by mdevaev