pikvm/pikvm #841: Annoying dialog box appears when heading back to home page of the device

n3gwg, 2022-11-02:
1) One goes to the home page of their PiKVM (let us say 192.168.1.1 is the IP).
2) Upon presentation of the menu the user clicks the "KVM" icon and is taken to the appropriate page.
3) The user then says, oh wait, I meant to go to click the "Terminal" icon! So, the user clicks the "<- PiKVM" icon in the upper left hand corner. Upon doing so the browser prompts with:

	"This page is asking you to confirm that you want to leave — information you’ve entered may not be saved. (Stay on page) (Leave page)".

4) The above entitled error becomes occurring notwithstanding the fact that the user has indeed made zero "unsaved changes" and should otherwise not be subject to having to deal with such a warning. No such warning is show when the user clicks the "Terminal" or "KVM" icon to get to those pages. Why?

The expected behavior would be for the browser to display the home page absent any such warning, just as no such warning becomes visible whence a user clicks on the "KVM" icon in the first place.

OS: Kali Linux
Browser: Firefox ESR (current version)

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: V3 (fully assembled unit from PiShop)
  - Video capture type: internal to V3 PiKVM as described herein above
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.157-1
kvmd-fan 0.22-1
kvmd-oled 0.13-1
kvmd-platform-v3-hdmi-rpi4 3.157-1
kvmd-webterm 0.42-1
  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 5.25-1
  - Linux kernel: `uname -a`
Linux pikvm1.stuart-lan 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

mdevaev, 2022-11-02:
This is done for accidental protection from pressing Alt+F4 or Ctrl+W. Since these hotkeys are not transmitted to the host due to browser restrictions, these hotkeys will cause your tab or browser to close.

If you don't like this behavior, just disable it:

[image]

closed 2022-11-02 by mdevaev

n3gwg, 2022-11-02:
Max,

Thank you for your most brisk and thorough response. Indeed, I do find it annoying and as such have disabled it! :)

I appreciate that you made this an option that a user can turn on or off, that is the right kind of thinking in my view for a developer to use. Do not force choices upon users, let them pick what they like (as I did and may change it later if I prefer a different choice).

I also wanted to say that I appreciate the way in which you responded with such detail and clarity using screenshots, red arrows, etc... very nice!

I really do like the PiKVM and despite having waited a long to to receive it, I regret nothing about the purchase whatsoever! :)

I will also state that I did mention this in the chat (and was told there was no reason to do it), but having the ability to obtain a PDF of the entire documentation set would be really nice.

Why you ask?
(What a good question on your part!)

1) A user (such as I) can print it out and leave a copy in their bathroom where it will get fully read over time (I do not take phones, tablets or other electronics in my bathroom)! :)

2) If it is in PDF format one can ostensibly easily search it and as well can access it without being online. I could put the PDF on my iPhone and read when I am flying or the like (as well as searching it then too).

3) A printed PDF manual looks super-dooper professional (with a table of contents, index, glossary, page numbers, etc...) and as everything else about the PiKVM is done professionally, why not this too?

All that being said, I realize that resources (people to work on it) may well be a portion of why it is currently not done and so I am not blind to such concerns either.

Thanks again for all your assistance.

Stuart
N3GWG