pikvm/pikvm #842: Add time/date stamp to drive -> image selection drop down menu

n3gwg, 2022-11-03:
When one uses the Drive menu and wishes to mount an image they click the down arrow and are presented the image selection menu inclusive of the size of the image. How difficult would it be to also have the date/time stamp of ISO in that window too? This would allow the user to know if they needed to update the image perhaps in a very convenient manner.

mdevaev, 2022-11-04:
It's not difficult but do you really need this? This is a rare case. Why just don't use date and time in the image name? like `foobar-20221104.iso`

n3gwg, 2022-11-04:
I have a tendency to give the ISOs the same name I am familiar with seeing them have when I download them from their source website. Moreover, image names can be quite long to begin with (like: "debian-live-11.4.0-amd64-gnome+nonfree.iso" or that kind of thing).

It is surely not an urgent feature request, I can live without it. However, the next time you are in that section of the code and you can get to implementing it, I would surely like to see it added. I understand that there may be functional bugs taking priority over modest improvements. I also think it would not require very much more screen real estate to do so as well.

mdevaev, 2022-11-05:
It takes too much space:

[image]

n3gwg, 2022-11-05:
Well I would suggest that of the entirety of the date/time group, the seconds and the portion thereafter could be disposed of. My reasoning is that most websites showing ISO downloads only depict the date and the time contains only hours and minutes anyway.

mdevaev, 2022-11-10:
Will be available in the next release.

closed 2022-11-10 by mdevaev

n3gwg, 2022-11-10:
That is most excellent and I appreciate in no uncertain terms your most brisk and open minded outlook with respect to the software component of the PiKVM. I think it will be a useful and utilitarian improvement.

Thank you again so very much.

n3gwg, 2022-11-14:
I am very happy with the way the changes work and thank you again.
You can close out this ticket if you wish.

Stuart