pikvm/pikvm #843: Right shift issue

Marcin22222, 2022-11-05:
**Describe the bug**
 Tested with few laptops ,windows 10. When you use right shift it stay in pressed state (behaves like caps lock). So all letters are big and when you press number special character appears. To work normally you have to use left shift which is not comfortable

**To Reproduce**
Steps to reproduce the behavior, like:
1. Log to kvm 
2. Start typing, and use right shift in the middle. From that moment it behaves like described above.

**Expected behavior**
Right shift should behave like left shift which mean normal shift beh. It should generate uppercase and special characters only when pushed, not as some kind of switch 

**Desktop (please complete the following information):**
 - OS: windows 10
 - Browser chrome, edge

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

mdevaev, 2022-11-05:
Sup. did you try it on linux or mac?

mdevaev, 2022-11-05:
I tried it on my win10+chrome laptop but I can't reproduce it.

Also please check on firefox.

arch1mede, 2022-11-05:
Also unable to repo this. Also please note, EDGE is unsupported. Please try different browsers, try private or incog window within those other browsers as the dev suggests.

Marcin22222, 2022-11-06:
You are right, it works on normal laptops. The issue Im observing shows when connecting from vm. So I assume that there are many variables in this equation in this situation.

mdevaev, 2022-11-06:
Are you using a browser inside a VM on a laptop, am I right? Which VM are you using? Maybe this is a known VM-specific problem.

Marcin22222, 2022-11-07:
Yes, it is azure vm.

mdevaev, 2022-11-07:
Try the keyboard tester in the browser and send a screenshot: https://unixpapa.com/js/testkey.html

mdevaev, 2022-11-10:
Okay, this is a VM problem, there's nothing we can do here. The browser cannot distinguish whether it is running on real hardware or not, and if the VM transmits incorrect key events to it, then there is no way to fix it.

closed 2022-11-10 by mdevaev