pikvm/pikvm #876: kvmd-oled.service not starting

nicedevil007, 2022-11-22:
**Describe the bug**

I followed the guide here => https://docs.pikvm.org/v3/#basic-setup
And on step 4 I tryed to enable the OLED with the following commands:

```
# rw
# systemctl enable --now kvmd-oled kvmd-oled-reboot kvmd-oled-shutdown
# systemctl enable --now kvmd-fan
# ro
```

It shows an error message that tells me I have to take a look in journalctl.
Here is the output:

```
[root@pikvm ~]# journalctl -xeu kvmd-oled.service
Nov 22 10:19:08 pikvm kvmd-oled[3771]:   File "/usr/lib/python3.10/site-packages/luma/oled/device/__init__.py", line 176, in __init__
Nov 22 10:19:08 pikvm kvmd-oled[3771]:     self.command(
Nov 22 10:19:08 pikvm kvmd-oled[3771]:   File "/usr/lib/python3.10/site-packages/luma/core/device.py", line 48, in command
Nov 22 10:19:08 pikvm kvmd-oled[3771]:     self._serial_interface.command(*cmd)
Nov 22 10:19:08 pikvm kvmd-oled[3771]:   File "/usr/lib/python3.10/site-packages/luma/core/interface/serial.py", line 101, in command
Nov 22 10:19:08 pikvm kvmd-oled[3771]:     raise luma.core.error.DeviceNotFoundError(
Nov 22 10:19:08 pikvm kvmd-oled[3771]: luma.core.error.DeviceNotFoundError: I2C device not found on address: 0x3C
Nov 22 10:19:08 pikvm systemd[1]: kvmd-oled.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- An ExecStartPre= process belonging to unit kvmd-oled.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Nov 22 10:19:08 pikvm systemd[1]: kvmd-oled.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- The unit kvmd-oled.service has entered the 'failed' state with result 'exit-code'.
Nov 22 10:19:08 pikvm systemd[1]: Failed to start Pi-KVM - Small OLED daemon.
-- Subject: A start job for unit kvmd-oled.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- A start job for unit kvmd-oled.service has finished with a failure.
--
-- The job identifier is 2672 and the job result is failed.
```

**Expected behavior**
OLED will start.

**PiKVM info:**
  - Raspberry Pi board version RPi 4 - 1GB
  - PiKVM platform v3
  - Video capture type [e.g. CSI bridge]
  - KVMD version:
```
kvmd 3.180-1
kvmd-fan 0.22-1
kvmd-oled 0.13-1
kvmd-platform-v3-hdmi-rpi4 3.180-1
kvmd-webterm 0.42-1
```
  - uStreamer version: 5.31-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
I already tryed a complete fresh installation with newly downloaded image. I also did the Update process for all packages before testing the OLED again.

mdevaev, 2022-11-22:
Please show `/boot/config.txt`.

nicedevil007, 2022-11-22:
```
[root@pikvm ~]# cat /boot/config.txt
# See /boot/overlays/README for all available options
initramfs initramfs-linux.img followkernel

hdmi_force_hotplug=1
gpu_mem=128
enable_uart=1
dtoverlay=tc358743
dtoverlay=disable-bt
dtoverlay=dwc2,dr_mode=peripheral
dtparam=act_led_gpio=13

# HDMI audio capture
dtoverlay=tc358743-audio

# SPI (AUM)
dtoverlay=spi0-1cs

# I2C (display)
dtparam=i2c_arm=on

# Clock
dtoverlay=i2c-rtc,pcf8563
```

mdevaev, 2022-11-22:
Could you take a photo of V3 HAT with LCD?

nicedevil007, 2022-11-22:
later :D right now I have a session on it with support. will report back

mdevaev, 2022-11-22:
Okay.

nicedevil007, 2022-11-22:
So here we go:

[image]

[image]

[image]

[image]

mdevaev, 2022-11-22:
Looks fine to me. Please try to flash the new OS image: https://files.pikvm.org/images/v3-hdmi-rpi4-box-latest.img.xz

nicedevil007, 2022-11-22:
> Looks fine to me. Please try to flash the new OS image: https://files.pikvm.org/images/v3-hdmi-rpi4-box-latest.img.xz

I did this as written above.

mdevaev, 2022-11-22:
Sorry, I'm not thinking straight because of COVID :/ Could you please contact our discord with a link here? Write that I advised you to do this. Our support specialists will help you.

Perhaps you have a defective screen, but to make sure of this, we need our HW guy to look at it.

nicedevil007, 2022-11-22:
I have a 2nd device that I baked on kickstarter a year ago or something like this, I can take this OLED and test it on the 2nd device and if this will not work as well, I can contact discord community :) will report back tomorrow.

mdevaev, 2022-11-22:
:ok_hand:

nicedevil007, 2022-11-23:
Ok, tested with 2nd display, that works out of the box.
So I bought a defect display. I will contact the reseller.

closed 2022-11-23 by nicedevil007