pikvm/pikvm #877: Add 4th and 5th mouse buttons (up, down)

gusuru, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
4th and 5th physical mouse buttons (up, down) do not respond when pressed.

**Describe the solution you'd like**
Add these buttons to window-stream.pug.

**Describe alternatives you've considered**

**Additional context**

gusuru, 2022-11-22:
**Desktop (please complete the following information):**

OS: Windows10
Browser: chrome
Version: 107.0.5304.107
VNC client not using

**PiKVM info:**

Raspberry Pi board version: RPi 4
PiKVM platform: v3-hdmi
Video capture type: CSI bridge
KVMD version: 3.173-1
uStreamer version: 5.31-1
Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH # 1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

keypad.js
```
	self.emitByCode = function(code, state, apply_fixes=true) {
		if (code in __merged) {
```
In my environment, it seems that the processing ends without finding up or down here.

mdevaev, 2022-12-04:
Mouse buttons are handled directly here: https://github.com/pikvm/kvmd/blob/master/web/share/js/kvm/mouse.js#L170

Try checking button events using debug mode. Open the JS console and use this URL: `https://pikvm/kvm/?debug=1`