pikvm/pikvm #878: Add Japanese keyboard keys

gusuru, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
The following physical keys on the Japanese keyboard do not respond when pressed.

```
IntlRo,108,0x87,reg:0x51,0x73,
KanaMode,109,0x88,reg:0x13,0x70,
Convert,110,0x8a,reg:0x64,0x79,
NonConvert,111,0x8b,reg:0x67,0x7b,
```
"IntlRo" is required to enter "_" (underline).
Other keys are used when entering Japanese characters.

**Describe the solution you'd like**
Add these keys to keymap.csv and window-keyboard.pug.

**Describe alternatives you've considered**
Personally, I only need to be able to input with a physical keyboard, and the priority of Virtual Keyboard is low.

**Additional context**
Japanese "106" Keyboard Layout
https://www.w3.org/TR/uievents-code/#keyboard-106

mdevaev, 2022-12-10:
Okay, I've done. Please update OS and test it.

PS: Should Hana, NonConvert and Convert be modifiers on the virtual keyboard?

gusuru, 2022-12-11:
Thank you.
"IntlRo", "KanaMode", "Convert", "NonConvert" could be entered.

I found out that I can use it, but with the Japanese keyboard, both Chrome and FireFox,
Even if I press the "KanaMode" physical key and release it immediately, it will remain pressed.

Also, when I checked the other keys, I found that "Backquote" and "CapsLock" would remain pressed even if I pressed and released them immediately.

sorry.
It seems to be a local specification of the Japanese keyboard.
https://bugs.chromium.org/p/chromium/issues/detail?id=612694

In the following cases, keyUp does not occur.
```
event.code == "Backquote" && ["Hankaku", "Zenkaku"].includes(event.key)) ||
event.code == "CapsLock" && ["Alphanumeric"].includes(event.key)) ||
event.code == "KanaMode" && ["Hiragana", "Katakana", "Romaji", "KanaMode"].includes(event.key))
```

If you don't mind, I would appreciate it if you could send me an additional keyUp.

https://github.com/pikvm/kvmd/blob/c551a08a91e45b878d63f306aa0d59fd64b327b4/web/share/js/keypad.js#L97-L113

```
	self.emitByKeyEvent = function(event, state) {
+		let isJpModeChangeKey = function(event) {
+			return (event.code == "Backquote" && ["Hankaku", "Zenkaku"].includes(event.key))
+				|| (event.code == "CapsLock" && ["Alphanumeric"].includes(event.key))
+				|| (event.code == "KanaMode" && ["Hiragana", "Katakana", "Romaji", "KanaMode"].includes(event.key));
+		};
+
-		if (event.repeat) {
+		if (event.repeat && !(__apply_fixes && isJpModeChangeKey(event))) {
			return;
		}

		let code = event.code;
		if (__apply_fixes) {
			// https://github.com/pikvm/pikvm/issues/819
			if (code == "IntlBackslash" && ["`", "~"].includes(event.key)) {
				code = "Backquote";
			} else if (code == "Backquote" && ["§", "±"].includes(event.key)) {
				code = "IntlBackslash";
			}
		}

		self.emitByCode(code, state);
+
+		if (__apply_fixes) {
+			if (state && isJpModeChangeKey(event)) {
+				self.emitByCode(code, false);
+			}
+		}
	};
```

> PS: Should Hana, NonConvert and Convert be modifiers on the virtual keyboard?

"KanaMode", "NonConvert", "Convert" key is not a modifier key.

gusuru, 2022-12-12:
Supplement.

When I display PiKVM's VirtualKeyboard in Chrome and click the "Backquote" or "CapsLock" button, it works as expected.
There is no key holding down on the remote side.

View the following html in Chrome and attach the log of pressing the physical keys.
```
<script>
document.addEventListener('keydown', (event)=>{
    console.log(new Date().toLocaleString('sv-SE'),'(keydown)','key:',event.key,'code:',event.code);
});
document.addEventListener('keyup', (event)=>{
    console.log(new Date().toLocaleString('sv-SE'),'(keyup)  ','key:',event.key,'code:',event.code);
});
</script>
```
Press and release the following keys in sequence.
```
"Backquote", "Space", "Backquote", "Space",
"CapsLock", "Space", "CapsLock", "Space",
"KanaMode", "Space", "KanaMode", "Space",
"Shift+KanaMode", "Space", "Shift+KanaMode", "Space",
"Alt+KanaMode", "Space", "Al+KanaMode", "Space"
```
[image]

Even though the key is pressed and released immediately, events occur in the order of keyup (previous mode) and keydown (switched  mode).

If I press the same key in succession, events will occur in the order keyup (previous mode), keydown (switched mode), and keydown (switched mode).

"Backquote", "CapsLock", and "KanaMode" on the Japanese keyboard are keys that switch the input mode, so there is no need to hold them down, but may press the same key twice in succession.

The logs are from Chrome, but are similar for FireFox.

mdevaev, 2022-12-13:
Pretty confusing logic. If you don't mind, I would ask you to make a patch and send a pull request.

mdevaev, 2022-12-21:
Sup?

mdevaev, 2023-01-22:
Sup