pikvm/pikvm #880: Mouse pointer does not move to the right edge of the screen

gusuru, 2022-11-22:
**Describe the bug**
Mouse pointer does not move to the right edge of the screen.

/usr/share/kvmd/web/share/js/kvm/mouse.js
```
    var __sendPlannedMove = function() {
        if (__absolute) {
            let pos = __planned_pos;
            if (pos.x !== __sent_pos.x || pos.y !== __sent_pos.y) {
                let geo = __getGeometry();
                let to = {
                    x: tools.remap(pos.x, geo.x, geo.width, -32768, 32767),
                    y: tools.remap(pos.y, geo.y, geo.height, -32768, 32767),
```
In 1920x1080 fullscreen mode, pos.x ranges from 0 to 1919, geo.x is 0, and geo.width is 1920.
When move the mouse pointer to the right edge : 1919 / 1920 * 65535 - 32768 = 32732.8671875 (< 32768)
I think that it is good to calculate with geo.width-1.

**To Reproduce**
Snap right with mouse on Windows 10.
https://support.microsoft.com/en-us/windows/snap-your-windows-885a9b1e-a983-a3b1-16cd-c531795e6241

**Expected behavior**
Can't snap right with mouse.

**Screenshots**
sorry no screenshots

**Desktop (please complete the following information):**
 - OS: Windows10
 - Browser: chrome
 - Version: 107.0.5304.107
 - VNC client not using

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: v3-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 3.173-1
  - uStreamer version: 5.31-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH # 1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**

adamoutler, 2023-04-24:
I can't reproduce this.  It works for me on Windows 10.

mdevaev, 2025-05-12:
I noticed it and rechecked. The coordinate mapping formulas were not quite correct. Fixed now. It will be available in the next release.

mdevaev, 2025-05-18:
Released in 4.72, please update with `pikvm-update`.

closed 2025-05-18 by mdevaev