pikvm/pikvm #882: IP Settings option on boot partition or in WebGui

tompointexe, 2022-11-26:
**Is your feature request related to a problem? Please describe.**
 Looking and linking to Issue (question) #82 here
The user was looking to get a static ip to PIKVM, and i think, for piKVM to be used in more professional environements, having this would be awesome

**Describe the solution you'd like**

A simple menu (or command / script) to setup static ip in the web gui, or an ip config file in the SDCard boot partition

**Describe alternatives you've considered**
Setting up the IP directly by hand in the tty can be done, but in a datacenter environement, having access to a mini hdmi to hdmi / VGA is not realy easy !
And DHCP is clearly not present here

**Additional context**
Using PiKVM's in datacenter environements were servers and their ipmi / KVM's uses all static ip's 

I know PiKVM is design primarely for HomeLab users, but it's a great IPMI to use in the datacenter world were recent motherboards (Looking at AM5 RN for new servers (game and VPS Hosting)) doesen't have a pro / datacenter version, and thoes boards are EXPENCIVE AF ! 

ps: Sorry for aproximative engilsh, French here without a translator 😅

mdevaev, 2022-12-04:
Sup. Configuring via pikvm.txt will be available in the next release. Two examples:
```bash
ETH_ADDR=x.x.x.x/xx
ETH_GW=x.x.x.x
ETH_DNS=x.x.x.x
```
Or
```bash
ETH_DHCP=1
```

closed 2022-12-04 by mdevaev

tompointexe, 2022-12-04:
Awesome !! Thanks a lot !