pikvm/pikvm #886: failed to create symbolic link Read-only file system

connected201, 2022-11-29:
hi,i want to create symlink, but i unable to do this
`rw` not help
```
root@pikvm kvmd-webterm]# rw
[root@pikvm kvmd-webterm]# ln -s /home/kvmd-webterm/share /var/lib/kvmd/msd/images/
ln: failed to create symbolic link '/var/lib/kvmd/msd/images/share': Read-only file system
```
how to fix? thanks.

mdevaev, 2022-11-29:
Don't do this. This storage is controlled by KVMD. In addition, if you add a link, it will give absolutely nothing, you will not get additional images.

connected201, 2022-11-29:
ok, then where i can change default path where is saved uploaded images? from `/var/lib/kvmd/msd/images/` to `/home/kvmd-webterm/share`.

mdevaev, 2022-12-01:
The path where you save the images should be able to be remounted in read-write mode. That is, to be a separate section. If you just change the path, you will break everything. What do you want to do?

connected201, 2022-12-01:
i want to mount remote share to use iso files, img files and other, to do not upload files in pikvm.
i want to use remote .img file that i can mount in readwrite, write info, and disconnect, without needing manual upload img file to remote server for extract files or other actions.

mdevaev, 2022-12-02:
I'm working on it in #100 . It will be available soon with some instructions. Just wait until it will be done.

closed 2022-12-02 by mdevaev