pikvm/pikvm #887: pikvm not working in bromite browser

connected201, 2022-11-30:
hi, i using on android devices bromite browser:  https://github.com/bromite/bromite
but pikvm not working on this browser :(
**bromite browser:**
[image]
**chrome browser:**
[image]

mdevaev, 2022-12-03:
I see that a lot of things are disabled in this browser. I don't have Android on hand to debug this. Most likely the reason is that they broke something from the JS side. If you can get debugging output from the browser's JS console, I'll take a look at it and MAYBE be able to fix it: https://developer.chrome.com/docs/devtools/

mdevaev, 2022-12-05:
Sup?

connected201, 2022-12-06:
hi, mobile version don't have devtools like in desktop version, maybe i need to connect over ADB on android, i will try in a few days to solve.

connected201, 2022-12-08:
strange things, today i opened pikvm in bromite and all work well. topic closed.

closed 2022-12-08 by connected201