pikvm/pikvm #891: rpi3 v0-hdmiusb - No Signal on the UI

raghu-mp, 2022-12-12:
Hi, 

I compiled the os using instructions at https://docs.pikvm.org/building_os/  Connected the HDMI USB capture device https://www.amazon.com/dp/B00KIALPWE?psc=1&ref=ppx_yo2ov_dt_b_product_details and seeing NO SIGNAL on the UI. kvmd-video is not showing up under /dev. Here are some details. Can someone help please?

**v4l2-ctl --list-dev**
UVC Camera (534d:2109): USB Vid (usb-3f980000.usb-1.4):
	/dev/video0
	/dev/video1
	/dev/media0

**v4l2-ctl --log-status**
VIDIOC_LOG_STATUS: failed: Inappropriate ioctl for device

**lsusb**
Bus 001 Device 007: ID 534d:2109 MacroSilicon 
Bus 001 Device 005: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**cat /etc/udev/rules.d/99-kvmd.rules** 
# https://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name
# https://wiki.archlinux.org/index.php/Udev#Setting_static_device_names
KERNEL=="video[0-9]*", SUBSYSTEM=="video4linux",KERNELS=="1-1.5:1.0", PROGRAM="/usr/bin/kvmd-udev-hdmiusb-check rpi3 %b", ATTR{index}=="0", GROUP="kvmd", SYMLINK+="kvmd-video"
KERNEL=="ttyAMA0", SYMLINK+="kvmd-hid"

**ls -l /dev/kvmd***    
lrwxrwxrwx 1 root root 7 Dec  9 20:08 /dev/kvmd-hid -> ttyAMA0

**v4l2-ctl --all**
Driver Info:
	Driver name      : uvcvideo
	Card type        : UVC Camera (534d:2109): USB Vid
	Bus info         : usb-3f980000.usb-1.4
	Driver version   : 5.15.68
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : UVC Camera (534d:2109): USB Vid
	Serial           : 
	Bus info         : usb-3f980000.usb-1.4
	Media version    : 5.15.68
	Hardware revision: 0x00002100 (8448)
	Driver version   : 5.15.68
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : UVC Camera (534d:2109): USB Vid
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x0200000d: from remote pad 0x100000a of entity 'Processing 2' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 4147200
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1920, Height 1080
	Default     : Left 0, Top 0, Width 1920, Height 1080
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-128 max=127 step=1 default=-11 value=-11
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=148 value=148
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=180 value=180
                            hue 0x00980903 (int)    : min=-128 max=127 step=1 default=0 value=0

mdevaev, 2022-12-13:
Hello. Try to change file ` /etc/udev/rules.d/99-kvmd.rules`
Line:
```
KERNEL=="video[0-9]*", SUBSYSTEM=="video4linux",KERNELS=="1-1.5:1.0", PROGRAM="/usr/bin/kvmd-udev-hdmiusb-check rpi3 %b", ATTR{index}=="0", GROUP="kvmd", SYMLINK+="kvmd-video"
```
To:
```
KERNEL=="video0", SUBSYSTEM=="video4linux", GROUP="kvmd", SYMLINK+="kvmd-video"
```

raghu-mp, 2022-12-13:
Thank you so much for the quick response ! It turns out the issue was the USB port on the rpi3. Plugged the HDMI USB dongle into another port and it started working ! Found this out while flashing the HID to the Arduino which was failing. After moving it to another USB port it worked. Then i tried the same USB port for the HDMI dongle as well and voila :)

Now onto figuring out how to make the Arduino + PL2303 Serial USB (appears as ttyUSB0) working as HID. I am referring to the setup at https://github.com/purplepalmdash/pikvmx86 and applying the same to rpi3 instead of the more complex HID setup with logical lever shifter etc. https://ku.nz/blog/ipmi.html also confirms it should work with KVMD. 

Any suggestions as to the changes required on rpi3 for this setup to work?

mdevaev, 2022-12-15:
It seems it should be enough to change the port in `platformio-avr.ini` and kvmd settings using `/etc/kvmd/override.yaml` (see `kvmd -m`).

mdevaev, 2022-12-15:
Yes

raghu-mp, 2022-12-15:
Getting the following error while flashing hid. 

Configuring upload protocol...
AVAILABLE: avr109
CURRENT: upload_protocol = avr109
Looking for upload port...
Using manually specified: /dev/ttyUSB0
Forcing reset using 1200bps open/close on port /dev/ttyUSB0
Waiting for the new upload port...
Uploading .pio/build/serial/firmware.hex

Connecting to programmer: .
Found programmer: Id = "\x8c\x\xac\x8"; type = \x88
    Software Version = \x84; Hardware Version =.\xac
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
avrdude: error: programmer did not respond to command: exit bootloader

avrdude done.  Thank you.

*** [upload] Error 1
=========================================================== [FAILED] Took 15.15 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
serial         FAILED    00:00:15.155
====================================================== 1 failed, 0 succeeded in 00:00:15.155 ======================================================
make: *** [Makefile:25: upload] Error 1

raghu-mp, 2022-12-16:
I was able to flash using upload_port = /dev/ttyACM0. Then connected it back via USB-TTL (/dev/ttlUSB0). Both video and keyboard are now working !! Thank you for the continued support!  You are AWSOME :)

closed 2022-12-16 by mdevaev