pikvm/pikvm #899: ipmi sol support stutters

awilliam, 2022-12-19:
**Describe the bug**
Using ipmi sol support, the serial output stutters.  For instance when viewing the console of a Linux system during boot or reviewing boot logs with dmesg, I'll get anywhere from a few lines to a few screen fulls of text before a brief pause and the output continues.  This is disruptive to use as a serial console server, which should be able to smoothly stream the console log, and as found on other hardware BMC devices or when directly using a terminal program such as minicom.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Configure IPMI SOL support as described in https://docs.pikvm.org/ipmi/#ipmi-bmc
2. Use a USB serial adapter, such as https://www.amazon.com/dp/B075YHFMC7
3. Note, add kvmd-ipmi to the uucp group to access USB serial (ex. gpasswd -a kvmd-ipmi uucp)
4. Configure all of the above at 115200 baud rate and connect to Linux host system configured for the same.
5. Use ipmitool on the pikvm system to connect to SOL: ex. ipmitool -I lanplus -U admin -P admin -H localhost sol activate
6. Run 'dmesg' from the SOL session

**Expected behavior**
Text should stream smoothly.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Fedora Linux
 - Browser: N/A
 - Version:N/A
 - VNC client: N/A

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]: RPI4
  - PiKVM platform [e.g. v2-hdmi]: v2-hdmiusb
  - Video capture type [e.g. CSI bridge]: USB
  - KVMD version: `pacman -Q | grep kvmd`: kvmd 3.173-1, kvmd-fan 0.22-1, kvmd-oled 0.13-1, kvmd-platform-v2-hdmiusb-rpi4 3.173-1, kvmd-webterm 0.42-1
  - uStreamer version: `pacman -Q | grep ustreamer`: ustreamer 5.31-1
  - Linux kernel: `uname -a`: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Maxim kindly logged into my pikvm system to see the issue first hand and asked me to file this issue.  Thanks!

mdevaev, 2023-03-26:
Could you update os and try ipmi sol again?

Also which package do you use for conserver? A can build it and add to the repo.

mdevaev, 2023-03-30:
@awilliam boop

awilliam, 2023-03-30:
Yes, that's the correct conserver package, seems there's no update relative to lack of armv7h builds.  Updated my pikvm and temporarily re-enabled ipmi sol, no obvious improvement from previous experience, slow, choppy output, no comparison to conserver.