pikvm/pikvm #903: Nginx not starting after update

GrafDe, 2022-12-26:
**Describe the bug**
I haven't updated the system in about 2 years. i upgraded the system from the old system.
I was doing an Update with:
rw
pacman -Syy
pacman -Su
reboot
After the Reboot the Webinterface shows just 500 Internal Server Error

when I restart the pikvm I get the error:

```
journalctl -xe
Dec 26 16:36:46 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
Dec 26 16:36:46 pikvm kvmd-cleanup[889]: ConfigError: Can't read config file '/etc/kvmd/main.yaml':
Dec 26 16:36:46 pikvm kvmd-cleanup[889]: ValueError: Invalid YAML in the file '/etc/kvmd/main.yaml':
Dec 26 16:36:46 pikvm kvmd-cleanup[889]: ValueError: Invalid YAML in the file '/etc/kvmd/override.yaml':
Dec 26 16:36:46 pikvm kvmd-cleanup[889]: ScannerError: while scanning for the next token
Dec 26 16:36:46 pikvm kvmd-cleanup[889]: found character '\t' that cannot start any token
Dec 26 16:36:46 pikvm kvmd-cleanup[889]:   in "/etc/kvmd/override.yaml", line 28, column 1
Dec 26 16:36:46 pikvm systemd[1]: kvmd.service: Control process exited, code=exited, status=1/FAILURE
Dec 26 16:36:46 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
Dec 26 16:36:46 pikvm systemd[1]: kvmd.service: Consumed 2.078s CPU time.
Dec 26 16:36:50 pikvm systemd[1]: kvmd.service: Scheduled restart job, restart counter is at 12.
Dec 26 16:36:50 pikvm systemd[1]: Stopped PiKVM - The main daemon.
Dec 26 16:36:50 pikvm systemd[1]: kvmd.service: Consumed 2.078s CPU time.
Dec 26 16:36:50 pikvm systemd[1]: Started PiKVM - The main daemon.
Dec 26 16:36:51 pikvm kvmd[890]: /usr/lib/python3.10/site-packages/kvmd/apps/kvmd/tesseract.py:80: RuntimeWarning: Can't load libtesseract: Can't find libtesseract
Dec 26 16:36:51 pikvm kvmd[890]:   warnings.warn(f"Can't load libtesseract: {err}", RuntimeWarning)
Dec 26 16:36:51 pikvm kvmd[890]: ConfigError: Can't read config file '/etc/kvmd/main.yaml':
Dec 26 16:36:51 pikvm kvmd[890]: ValueError: Invalid YAML in the file '/etc/kvmd/main.yaml':
Dec 26 16:36:51 pikvm kvmd[890]: ValueError: Invalid YAML in the file '/etc/kvmd/override.yaml':
Dec 26 16:36:51 pikvm kvmd[890]: ScannerError: while scanning for the next token
Dec 26 16:36:51 pikvm kvmd[890]: found character '\t' that cannot start any token
Dec 26 16:36:51 pikvm kvmd[890]:   in "/etc/kvmd/override.yaml", line 28, column 1
Dec 26 16:36:51 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
Dec 26 16:36:52 pikvm kvmd-cleanup[897]: ConfigError: Can't read config file '/etc/kvmd/main.yaml':
Dec 26 16:36:52 pikvm kvmd-cleanup[897]: ValueError: Invalid YAML in the file '/etc/kvmd/main.yaml':
Dec 26 16:36:52 pikvm kvmd-cleanup[897]: ValueError: Invalid YAML in the file '/etc/kvmd/override.yaml':
Dec 26 16:36:52 pikvm kvmd-cleanup[897]: ScannerError: while scanning for the next token
Dec 26 16:36:52 pikvm kvmd-cleanup[897]: found character '\t' that cannot start any token
Dec 26 16:36:52 pikvm kvmd-cleanup[897]:   in "/etc/kvmd/override.yaml", line 28, column 1
Dec 26 16:36:52 pikvm systemd[1]: kvmd.service: Control process exited, code=exited, status=1/FAILURE
Dec 26 16:36:52 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
Dec 26 16:36:52 pikvm systemd[1]: kvmd.service: Consumed 2.089s CPU time.
Dec 26 16:36:55 pikvm systemd[1]: kvmd.service: Scheduled restart job, restart counter is at 13.
Dec 26 16:36:55 pikvm systemd[1]: Stopped PiKVM - The main daemon.
Dec 26 16:36:55 pikvm systemd[1]: kvmd.service: Consumed 2.089s CPU time.
Dec 26 16:36:55 pikvm systemd[1]: Started PiKVM - The main daemon.
Dec 26 16:36:56 pikvm kvmd[898]: /usr/lib/python3.10/site-packages/kvmd/apps/kvmd/tesseract.py:80: RuntimeWarning: Can't load libtesseract: Can't find libtesseract
Dec 26 16:36:56 pikvm kvmd[898]:   warnings.warn(f"Can't load libtesseract: {err}", RuntimeWarning)
Dec 26 16:22:33 pikvm kvmd[1745]: ConfigError: Can't read config file '/etc/kvmd/main.yaml':
Dec 26 16:22:33 pikvm kvmd[1745]: ValueError: Invalid YAML in the file '/etc/kvmd/main.yaml':
Dec 26 16:22:33 pikvm kvmd[1745]: ValueError: Invalid YAML in the file '/etc/kvmd/override.yaml':
Dec 26 16:22:33 pikvm kvmd[1745]: ScannerError: while scanning for the next token
Dec 26 16:22:33 pikvm kvmd[1745]: found character '\t' that cannot start any token
Dec 26 16:22:33 pikvm kvmd[1745]:   in "/etc/kvmd/override.yaml", line 28, column 1
Dec 26 16:22:33 pikvm systemd[1]: kvmd.service: Main process exited, code=exited, status=1/FAILURE
Dec 26 16:22:34 pikvm kvmd-cleanup[1755]: ConfigError: Can't read config file '/etc/kvmd/main.yaml':
Dec 26 16:22:34 pikvm kvmd-cleanup[1755]: ValueError: Invalid YAML in the file '/etc/kvmd/main.yaml':
Dec 26 16:22:34 pikvm kvmd-cleanup[1755]: ValueError: Invalid YAML in the file '/etc/kvmd/override.yaml':
Dec 26 16:22:34 pikvm kvmd-cleanup[1755]: ScannerError: while scanning for the next token
Dec 26 16:22:34 pikvm kvmd-cleanup[1755]: found character '\t' that cannot start any token
Dec 26 16:22:34 pikvm kvmd-cleanup[1755]:   in "/etc/kvmd/override.yaml", line 28, column 1
Dec 26 16:22:34 pikvm systemd[1]: kvmd.service: Control process exited, code=exited, status=1/FAILURE
Dec 26 16:22:34 pikvm systemd[1]: kvmd.service: Failed with result 'exit-code'.
Dec 26 16:22:34 pikvm systemd[1]: kvmd.service: Consumed 2.083s CPU time.
```

what i need to do for resolve thise errors?

mdevaev, 2022-12-26:
Hello. Apparently you have modified `/etc/kvmd/override.yaml` and used tabs instead of four spaces as an indentation. Fix it and everything will work

GrafDe, 2022-12-26:
Yes you are right. wrote tab. I read carefully what the program writes in error messages and corrected them. now everything works.

GrafDe, 2022-12-26:
Thank you for fast help ))

closed 2022-12-26 by GrafDe

mdevaev, 2022-12-26:
:ok_hand: