pikvm/pikvm #906: Bluetooth mouse/keyboard not automatically connecting to remote computer (windows 10)

susadmin, 2023-01-05:
I have followed the instructions setout at: https://docs.pikvm.org/bluetooth_hid

Mouse and keyboard work perfectly through bluetooth on my remote windows 10 computer until I reboot the remote machine. After that, the PiKVM HID mouse and keyboard are inoperative until I unpair and then repair them again.

I have changed the windows Bluetooth Support Service to start Automatically instead of Manually (it was running anyway) but it has no effect. The PiKVM HID shows as 'paired' but not 'connected'. I have also tried pivkm web menu System -> Reset keyboard & mouse but this has no effect either.

I realise this is probably more of a Windoze problem but does anyone have any pointers?

My pikvm is fully updated running perfectly in every other way on a pi3.

Thanks for such a fantastic project!

mdevaev, 2023-03-07:
Hello. Sorry for the long answer. Unfortunately, I haven't had any experience using BT on Windows 10, so I don't know if there's any way to fix it :(

pssara, 2023-04-26:
Hi, I have the same issue. Can anybody help ?

susadmin, 2023-04-26:
Gave up in the end. It just would not work reliably. Probably more a windows problem rather than pikvm

mdevaev, 2023-04-26:
Sorry. It's strange that this works fine with iOS. If I find out something, I'll let you know.

closed 2023-04-26 by mdevaev

pssara, 2023-04-27:
Hi there. I am sorry, is it supposed to only work with Mac? 
I see that the file that is created under /var/lib/bluetooth/MAC/seeings has the discoverable flag to false. This file is created somehow. maybe Windows needs to have the device discoverable?

mdevaev, 2023-04-27:
It is supposed to work everywhere. The device does not need to be discoverable for this, this flag is needed to establish a connection for the first time. It occurs when you do a Reset HID.