pikvm/pikvm #908: Add signal switch to USB port to prevent voltage feedback

Slarag, 2023-01-13:
**Is your feature request related to a problem? Please describe.**
I'm a hardware verification engineer and use your project (or sometimes a standalone Raspberry Pi) for controlling some tests. One of them includes continous power cycling of the device under test (DUT). When the PiKVM (or Pi) is connected to the DUT via USB (the USB-C port where mouse/keyboard/etc are emulated) and power source is disconnected from the DUT, voltage feedback from the PiKVM/Pi to the DUT  through the D+/- lines can be observed. 

**Describe the solution you'd like**
The USB client port should be physically disconnected when no external 5V VBUS are supplied on the USB-C connector. This could be achieved e.g. by using a USB Highspeed switch like the Texas Instruments [TS3USB3000](https://www.google.com/search?q=TS3USB3000&rlz=1C1GCEU_deDE926DE926&oq=TS3USB3000&aqs=chrome.0.69i59j0i19i512j0i19i30l2j0i19i30i625j69i60l3.545j0j7&sourceid=chrome&ie=UTF-8) where the external VBUS voltage is used for switching the connection.

**Describe alternatives you've considered**
External USB module which includes the signal switch. This could possibly also be used with a native Raspberry Pi , but probably has higher costs due to the additionally needed USB connectors and separate PCB.

**Additional context**
Example wiring:
 - Connect D+/- to connector facing to Raspberry Pi
 - Connect USB+/- to USB-C connector
 - MHL+/- are not connected
 - Pull down on OE#
 - Pull up on SEL and a FET connected to pull the signal low when 5V are supplied from the USB-C connector

mdevaev, 2023-01-21:
Hello. Which device we are talking about? V2 DIY or V3?

Slarag, 2023-02-06:
Hello. Same behavior with a plain Raspberry Pi 4 and a PiKVM V3.
I think it is related to the USB speed identification logic of the SoC. In client mode, the Pi has to pull the D+ line to 3.3V with a 1k5 resistor.  ([see explanation](https://www-user.tu-chemnitz.de/~heha/hsn/chm/usb.chm/usb2.htm#:~:text=Speed%20Identification,as%20a%20full%20speed%20device.)).

The [USB 2.0 spec](https://www.usb.org/document-library/usb-20-specification) (document "usb_20.pdf" in the zip file, page 141, section 7.1.5.1) says:

> The voltage source on the pull-up resistor must be derived from or controlled by the power supplied on the USB
cable such that when VBUS is removed, the pull-up resistor does not supply current on the data line to which it is
attached.

This section also describes device speed identification in detail.

The pull up on the Pi (probably integrated in the SoC) is supplied with a voltage derived from the main power supply. Usually this is the 5V VBUS from the USB-C connector. This conforms to the spec.

But when the Pi is supplied through the 5V pins on the GPIO pin header, the 5V VBUS from the USB connector are no longer the main power source and the activation of the pull up is no longer in sync with the specification.

The solution I described above will emulate a physical disconnect of the USB connection to prevent this. The pull up (and the whole Pi) is disconnected from the host PC's USB data lanes when 5V VBUS are not applied.

mdevaev, 2023-03-21:
According to USB specification about self-powered devices (in case of OTG):

[image]

Host should handle this situation by itself. 

Besides, PiKVM use Raspberry Pi 4 without any modification, and SoC USB hardware stack implementation is out of scope. However - the solution you suggested by using USB multiplexer (just to be clear it is electrical solution, not physical wire disconnection) that is present in PiKVM hardware is possible. PiKVM V4 can handle it because V4 has some special functionality, but not V3.