pikvm/pikvm #910: Create and Expose "Power_State" Via API

NateroniPizza, 2023-01-14:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
If possible, I believe it would be a good idea to have a "power_state" or "status" item exposed via the API (https://docs.pikvm.org/api/) under https://<pikvm-ip>/api/atx.

Proposed states, and the logic that would accompany them (unless there is some other way you could think of to determine a computer's power state outside of gleaning it from the power LED):
On = Power LED on
Off = Power LED off
Sleep = Power LED flashing

I can currently get the power on and off state easily by checking whether the power LED is off or on. However, the only way to reliably determine whether a computer is in sleep mode would be to decrease the polling interval substantially to watch for the power LED flashing off-and-on, and design the logic in the request-originating application. If this logic could be handled locally on the PiKVM, and a "status" or "power_state" item added, then the polling interval could be left higher.

**Describe alternatives you've considered**
None that I have been able to find, other than greatly decreasing the polling interval of the application originating the state requests.

**Additional context**
n/a

mdevaev, 2023-01-18:
As far as I understand, this is specific to different motherboards. It's worth doing, but I'm not sure exactly how. I'll think about it.

NateroniPizza, 2023-01-23:
Sounds good. I ended up going with a smart plug to determine whether the PC is on or off/asleep, and will treat off andasleep the same, so isn't a pressing matter by any means. If this becomes available, though, I would definitely pull that data into HomeAssistant, to either replace or supplement the power draw, though.