pikvm/pikvm #911: Pi Zero 2 wifi issues

haggis-supper, 2023-01-14:
**Describe the bug**
Wifi is not getting an ip address.

**To Reproduce**
Download the latest v2-hdmi-zero2w-latest.img and write it using raspberry pi imager as a custom image.
Edit the pikvm.txt file and ensure it has the following lines
  FIRSTBOOT=1
  WIFI_ESSID="xxxxx"
  WIFI_PASSWD="xxxxx"

I've used 2 diffeerent zero 2 w boards bought at different times ( so not from the same batch )
- 1 board with CSI-Capture, i2c LCD and ATX board connected
- 1 board with nothing connected

When i plug in an HDMI screen and leave it running i can see the ip address showing up is 127.0.0.1 or 2.
Swapping the SD card to a Bastion image, wifi is picked up every time.

**Expected behavior**
The Output on HDMI should show 192.168.1.xx ( as per my networks DHCP offers )

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox
 - Version 108.0.2
 - Raspberry Pi Imager Version : v1.7.3

**PiKVM info:**
  - Raspberry Pi board version : PI zero2 - 2021
  - PiKVM platform : v2-hdmi
  - Video capture type : CSI bridge

haggis-supper, 2023-01-15:
Figured it out by disabling the virtual keyboard driver in config.txt and logging into the box.

It comes down to the wireless country definition, it needs set to GB in order to pick up the higher wireless channels( where my router is offering its SSID.

 # iw reg set GB

mdevaev, 2023-01-18:
Hello! You can pass `WIFI_REGDOM=GB` to `/boot/pikvm.txt`, everything should work with this. Since this apparently solves your problem, I will close this issue and reopen if necessary.

closed 2023-01-18 by mdevaev