pikvm/pikvm #925: User is experiencing strange issues with single-tap to single-click input with bVNC Pro on iOS with pikvm

iiordanov, 2023-01-28:
A user of bVNC Pro on iOS reported some strange behavior with pikvm whereas it seems the same taps work with xev and an Ubuntu system.

The issue that was opened with a great description is here:
https://github.com/iiordanov/remote-desktop-clients/issues/411

Are you able to replicate the behavior or could tell by looking what it is that's causing it and if there is a workaround?

Any input and suggestions are welcome. I am not in possession of a pikvm at the moment so I am unable to test this but maybe you have an Apple device and can give me some insight?

Thanks!!
iordan

mdevaev, 2023-01-29:
Hello. I'll check it.

c0dyhi11, 2023-02-28:
Hello,
Any update on this?

mdevaev, 2023-03-03:
I'll do it tomorrow.

mdevaev, 2023-03-07:
I found out what the problem is. bVNC sends the movement and click event in one message, and the PiKVM VNC proxy processes the click before the movement in the message. I changed the order to the opposite, now everything works as it should.

The fix will be released after few hours.

c0dyhi11, 2023-03-09:
Awesome! Thanks!!