pikvm/pikvm #926: Android user of bVNC reports issue with clipboard integration and pikvm

iiordanov, 2023-01-28:
A user is reporting that bVNC for Android is pasting the contents of his device's clipboard into his pikvm upon connecting.

A link to the review:
https://play.google.com/console/u/0/developers/8264949564453373715/app/4973068694465461699/user-feedback/review-details?reviewId=2f640e13-7b9f-4ca5-9cea-7a9d5e5e8602&corpus=PUBLIC_REVIEWS

I am unable to reproduce this behavior with other VNC servers and I do not have access to a pikvm at the moment - do you have any idea what could be causing this?

Thanks!
iordan

NiKiZe, 2023-01-28:
There is really no way to copy clipboard data into a device over USB, what has to happen is that the data is "typed out" if pikvm is doing this, or if it is something that happens in the client when clipboard support isn't available?

You should be able to run a docker image of pikvm and maybe do a packet trace there to see if you can reproduce the "clipboard transfer"?

iiordanov, 2023-01-28:
Hi Christian, So pikvm does not support VNC clipboard integration (client
but text / server cut text) at all?

mdevaev, 2023-01-29:
@iiordanov sup. Could you share that review? I can't open it.

At the moment, clipboard support is very limited because PiKVM is a hardware integration with PC, that is, we don't communicate with the OS. In PiKVM itself, image recognition from the screen is used for copying. The same goes for paste text - this is an imitation of the sequence of keystrokes.

iiordanov, 2023-01-29:
Hi @mdevaev, thanks for looking into this. The review:

> Got this to use with my Computer via PiKVM and works well for the most part. Though there 2 issues. 1. Mouse movement is fairly erratic when moving very slowly. This makes fine control nearly impossible. 2. MAJOR problem when first connecting. The phones clipboard is pasted into the remote machine as if being typed out by the keyboard. Executes shortcuts, types in chats, etc.. Thanks for quick reply, will report this using the link you provided. Thanks!

mdevaev, 2023-01-29:
Well, yes, that's exactly how it works. Unfortunately, this is a technical limitation.

mdevaev, 2023-01-29:
By the way, I would be interested if your client was one hundred percent compatible with our server. We can give you PiKVM V3 for free if you do H264 support and the rest of the pieces. What do you say?

iiordanov, 2023-01-30:
Hi Maxim, you've mentioned this encoding before, but I never saw (or don't
remember seeing) any documentation on it. Do you have something for me to
look at?

mdevaev, 2023-01-30:
Here, we're documented and officially registered it: https://github.com/rfbproto/rfbproto/blob/master/rfbproto.rst#open-h-264-encoding

mdevaev, 2023-03-07:
@iiordanov boop

iiordanov, 2023-03-11:
Thanks @mdevaev, I'll send this along to the user. Also when I find time, I'll work on these:

https://github.com/iiordanov/remote-desktop-clients/issues/441
https://github.com/iiordanov/remote-desktop-clients/issues/442

closed 2023-03-11 by iiordanov