pikvm/pikvm #932: Run PiKVM in container

yazdan, 2023-02-02:
Problem:
Running PiKVM on any other device is really hard and need you to compile and run whole OS image.

If container images was available like the approach `home-assistant` has followed in `HAOS` I think it would help
people like me who are not that familiar with building OS images etc. can run PiKVM on arbitrary hardware using cheap HDMI-to-USB dongle. It is possible to mount hardware devices in container, so I see no blocking issues. I don't actually know who to mount or access GPIO/SPI in the container, but there should be a way.

As far as I checked the repositories, we have couple of independent modules that are collaborating to provide functionality like `ustreamer`, `kvmd`.

If there is some documentation that depicts what are the components and how they interact, I can try to create a prototype for it.

arch1mede, 2023-02-08:
I am afraid this will never become reality, this will not ever be a container as per the [faq](https://docs.pikvm.org/faq/#common-questions) if it was easy, this would have been done already.

arch1mede, 2023-02-08:
https://github.com/pikvm/pikvm/issues/688
https://github.com/pikvm/pikvm/issues/486
https://github.com/pikvm/pikvm/issues/259

yazdan, 2023-02-08:
Thanks for your answers, 
I will check them then if have something will come back to you

closed 2023-02-08 by yazdan

arch1mede, 2023-11-27:
I formulated my previous remarks based on Docker's operation. I do not consider it advantageous to execute a PiKVM docker container on RPiOS or any distro due to numerous inoperative features. When utilizing an RPi SBC, it is preferable to obtain a pre-compiled image or utilize the options accessible in Discord's #unofficial_ports. This method alters RPiOS by integrating PiKVM fully, and the developer has also achieved functionality on x86 architecture.

While I do not agree with the docker image being made, I also understand why someone would want it, I was curious about it and found that your 2nd link and docker pull link both produced a 404 error, I would kindly ask that this be fixed.

Prototyped, 2023-11-27:
Ah, I'd accidentally left the container image private. That is now fixed.